don/BluetoothSerial #277: iOS Support for Microchip's RN-4677

sebj54, 2016-11-24:
Hi everyone,

I tried to add support for Microchip's RN-4677 on my own but it doesn't work at the moment. I don't know what I made wrong so I'm here to ask for help.

My Bluetooth module is this one: http://www.microchip.com/wwwproducts/en/RN4677
It supports both Classic Mode and BLE.

I didn't find the UART service UUID in the documentation but there are many and I don't really know where to find in all the documents.

I tried to run LightBlue app to find the UUID and edited `BLE.m` and `BLEDefines.h` files but it doesn't work so I Should have make a mistake in code or with the UUID.

Here is the screenshot from LightBlue app:
[image]

And here are the changes I made:

```diff
diff --git a/plugins/cordova-plugin-bluetooth-serial/src/ios/BLE.m b/plugins/cordova-plugin-bluetooth-serial/src/ios/BLE.m
index 9cc3e25..f05fe06 100644
--- a/plugins/cordova-plugin-bluetooth-serial/src/ios/BLE.m
+++ b/plugins/cordova-plugin-bluetooth-serial/src/ios/BLE.m
@@ -25,6 +25,7 @@ @implementation BLE
 static int rssi = 0;
 
 // TODO should have a configurable list of services
+CBUUID *microchipServiceUUID;
 CBUUID *redBearLabsServiceUUID;
 CBUUID *adafruitServiceUUID;
 CBUUID *lairdServiceUUID;
@@ -208,11 +209,12 @@ - (int) findBLEPeripherals:(int) timeout
     [NSTimer scheduledTimerWithTimeInterval:(float)timeout target:self selector:@selector(scanTimer:) userInfo:nil repeats:NO];
 
 #if TARGET_OS_IPHONE
+    microchipServiceUUID = [CBUUID UUIDWithString:@MICROCHIP_SERVICE_UUID];
     redBearLabsServiceUUID = [CBUUID UUIDWithString:@RBL_SERVICE_UUID];
     adafruitServiceUUID = [CBUUID UUIDWithString:@ADAFRUIT_SERVICE_UUID];
     lairdServiceUUID = [CBUUID UUIDWithString:@LAIRD_SERVICE_UUID];
     blueGigaServiceUUID = [CBUUID UUIDWithString:@BLUEGIGA_SERVICE_UUID];
-    NSArray *services = @[redBearLabsServiceUUID, adafruitServiceUUID, lairdServiceUUID, blueGigaServiceUUID];
+    NSArray *services = @[microchipServiceUUID, redBearLabsServiceUUID, adafruitServiceUUID, lairdServiceUUID, blueGigaServiceUUID];
     [self.CM scanForPeripheralsWithServices:services options: nil];
 #else
     [self.CM scanForPeripheralsWithServices:nil options:nil]; // Start scanning
@@ -525,7 +527,13 @@ - (void)peripheral:(CBPeripheral *)peripheral didDiscoverServices:(NSError *)err
         // Determine if we're connected to Red Bear Labs, Adafruit or Laird hardware
         for (CBService *service in peripheral.services) {
 
-            if ([service.UUID isEqual:redBearLabsServiceUUID]) {
+            if ([service.UUID isEqual:microchipServiceUUID]) {
+                NSLog(@"RedBearLabs Bluetooth");
+                serialServiceUUID = microchipServiceUUID;
+                readCharacteristicUUID = [CBUUID UUIDWithString:@MICROCHIP_CHAR_TX_UUID];
+                writeCharacteristicUUID = [CBUUID UUIDWithString:@MICROCHIP_CHAR_RX_UUID];
+                break;
+            } else if ([service.UUID isEqual:redBearLabsServiceUUID]) {
                 NSLog(@"RedBearLabs Bluetooth");
                 serialServiceUUID = redBearLabsServiceUUID;
                 readCharacteristicUUID = [CBUUID UUIDWithString:@RBL_CHAR_TX_UUID];
diff --git a/plugins/cordova-plugin-bluetooth-serial/src/ios/BLEDefines.h b/plugins/cordova-plugin-bluetooth-serial/src/ios/BLEDefines.h
index d852fb5..47b1a84 100644
--- a/plugins/cordova-plugin-bluetooth-serial/src/ios/BLEDefines.h
+++ b/plugins/cordova-plugin-bluetooth-serial/src/ios/BLEDefines.h
@@ -1,16 +1,21 @@
 
 /*
- 
+
  Copyright (c) 2013 RedBearLab
- 
+
  Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation files (the "Software"), to deal in the Software without restriction, including without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies of the Software, and to permit persons to whom the Software is furnished to do so, subject to the following conditions:
- 
+
  The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.
- 
+
  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
- 
+
  */
 
+// MICROCHIP RN4677 Service
+#define MICROCHIP_SERVICE_UUID                         "49535343-FE7D-4AE5-8FA9-9FAFD205E455"
+#define MICROCHIP_CHAR_TX_UUID                         "49535343-FE7D-4AE5-8FA9-9FAFD205E455"
+#define MICROCHIP_CHAR_RX_UUID                         "49535343-FE7D-4AE5-8FA9-9FAFD205E455"
+
 // BlueGiga Service
 #define BLUEGIGA_SERVICE_UUID                         "1D5688DE-866D-3AA4-EC46-A1BDDB37ECF6"
 #define BLUEGIGA_CHAR_TX_UUID                         "AF20fBAC-2518-4998-9AF7-AF42540731B3"

```

Thanks in advance for your help!

don, 2017-01-05:
@sebj54 it looks like you're close with the changes above. 

It's possible that you need to switch TX and RX. Some devices define these from the peripheral's perspective and others from the central's. It's also possible you used the wrong characteristics. I'm not sure which 2 of the 5 do what you need. It's tough to tell without the hardware or documentation.

You might need to implement flow control or credits to get this to work.  A ublox ConnectBlue dual mode radio required flow control https://github.com/don/ConnectBlue/blob/master/www/js/index.js

sebj54, 2017-01-05:
Thanks for your answer!

I tried several configurations and didn't make it work. So I decided to see what other developers do with BLE in Objective-C.
I ended up with a solution that scans for any services: https://github.com/medialta/BluetoothSerial/blob/master/src/ios/BLE.m#L221

What are flow control and credits? That might be what I missed.

don, 2017-01-06:
You need to scan for all services because the device isn't advertising the SPP service? Is the hardware advertisings some other service that you can scan for? (Expand the Advertising section in LightBlue)

Once you changed the scanning code does everything work?

closed 2017-03-17 by don