don/BluetoothSerial #278: Sending ESC/POS Commands

umair975, 2016-11-28:
How can I send ESC/POS Commands along with text

I am trying this.. the aim is to print an arabic string by chaiging the codepage but the printer is still printing garbage..

var data = new Uint8Array([3]);
bluetoothSerial.write(data, success, failure);

don, 2017-01-05:
You'll want to send the ESC/POS commands as bytes. To send 3 line feeds, you'd *probably* send something like this.

    var data = new Uint8Array([0x0A, 0x0A, 0x0A]);
    bluetoothSerial.write(data, success, failure);

Arabic strings are UTF-8 or UTF-16 so you're going to need to convert the string into a byte array and then send it to the printer.  (I'm just guessing here. You might need to do other things like set the correct font or language code to really get this to work.)

    function stringToBytes(string) {
        // based on http://ciaranj.blogspot.fr/2007/11/utf8-characters-encoding-in-javascript.html

        var bytes = [];

        for (var n = 0; n < string.length; n++) {

            var c = string.charCodeAt(n);

            if (c < 128) {

                bytes[bytes.length]= c;

            } else if((c > 127) && (c < 2048)) {

                bytes[bytes.length] = (c >> 6) | 192;
                bytes[bytes.length] = (c & 63) | 128;

            } else {

                bytes[bytes.length] = (c >> 12) | 224;
                bytes[bytes.length] = ((c >> 6) & 63) | 128;
                bytes[bytes.length] = (c & 63) | 128;

            }

        }

        return new Uint8Array(bytes);
    }

    var hello = "مرحبا";
    var data = stringToBytes(hello);
    bluetoothSerial.write(data, success, failure);

closed 2017-03-17 by don

rakesanug, 2017-11-09:
It is not working @don

riham-wazeer411, 2019-03-16:
@umair975 I'm trying to send Arabic String to OPOS Printer using PrintNormal so I kind of have the same issue, would like to ask if you solved this problem, please provide me with a clue