don/BluetoothSerial #283: Connect successfully but failed to write data

sy1016, 2016-12-21:
Hi Don,

Can this plugin work with PIC microcontroller (HC-06)?
I try with "Chat" sample and it's connected successfully, however, it failed when reading/write to data

Thanks.

sy1016, 2016-12-22:
Hi Don, i send command "AT" and it should response with "Ok" but then it failed even with line feed added.

bluetoothSerial.write("AT\n", success, failure);

it will response with any input i key in, if i key in "AT", it will response with "AT", if i enter "Hello" and send, the response came back as "Hello", which is incorrect

don, 2017-01-05:
@sy1016 some devices will accept AT commands, but it needs to be in mode that can accept them. Sometimes you need to send +++.  (You'll need to be using Android for this to work.)

Some devices might not like the strings, you could try encoding the data yourself into a byte array. See [bluetoothSerial.write](https://github.com/don/BluetoothSerial#write) for more info.

closed 2017-01-05 by don