don/BluetoothSerial #285: Unable to connect

nithinskariah, 2017-01-09:
Hi,

Im developing one Hybrid App using Ionic Framework (  Version 2 ) . In this app there is a functionality  to connect my app with a Bluetooth Keypad . I choose this plugin to make that possible .  

Plugin is able to detect this Keypad and listed in Paired and Unpaired list.  But when i try to connect device I'm getting ' Unable to connect error'. I try many times and after that I logged in Java code . Java console shows ' Socket timeout ' error .

My bluetooth device name is  ' Satechi Bluetooth Wireless Numeric Keypad '.

Following was my code to connect 
```

 var class_ob = this;
     BluetoothSerial.list()
     .then(function(data){console.log(data);
       data.forEach(function(device) {
          if(device.name == 'Bluetooth Keypad')
          {
            class_ob.bluetoothConnect(device.address);
            
          }
        })
      
      }, function(error){}); 

  bluetoothConnect(address)
  {
    console.log('bluetoothConnect',address);
    
    BluetoothSerial.connect(address)
          .subscribe(
            data=>{
              console.log('connect',data);
            },
            error =>
            {
              console.log('connect_error',error);
            }
          )
       
  }
```

Bluetooth keypad is not a BLE device . So I cant test it in IOS

Is there any wrong in my code   ?

diegozhu, 2017-01-24:
test first on android.  My code could work successfully on android but not working on iOS, always says that "Could not find peripheral" , I strongly recommend u test on android first.

nithinskariah, 2017-01-27:
im getting ' Unable to connect error' in Android

don, 2017-02-02:
@nithinskariah this plugin is for serial communication. You might be able to connect to the keyboard, but you can't get data since it won't support RFCOMM. A Bluetooth keyboard is a HID (Human Interface Device) so it should just work on iOS and Android after pairing. You shouldn't need this plugin.

closed 2017-02-02 by don