don/BluetoothSerial #286: Sequential connection to different devices results in erroneous read() results

lentyaishe, 2017-01-30:
Hi,

The plugin is great and works perfectly for my Android apps.

Yet, I've found some small issue that I think will be very easy to fix. In my app I'm connecting to different devices in sequence and get data from these devices while the data itself differs. The problem occurs when I'm doing the following sequence: `connect()` to device X => `read()` data continuously from this device => `disconnect()` from device X => `connect()` to device Y => `read()` data continuously from device Y. The problem is that the first `read()` on device Y brings the last data that was read on device X.

I suppose it could be easily fixed by just cleaning the read buffer in the `connect()` method in `BluetoothSerial.java`. It would be great if this issue could be fixed officially and for all available platforms.

Thanks!

don, 2017-02-02:
@lentyaishe disconnect should probably clear the buffer, or is should be cleared as part of the connection process.

For now, your app will need to clear the buffer. I'll look at building this into the plugin.

lentyaishe, 2017-02-03:
@don I've just put these two lines of code:
`int length = buffer.length();`
`buffer.delete(0, length);`
in the `connect()` method of `BluetoothSerial.java` after line #343 and it works like a charm but it would be great to have it officially in the plugin.
Thanks!

commit ba78b3b2 (2017-03-17): clear old data from buffer on new connection fixes #286
closed 2017-03-17 by don