don/BluetoothSerial #288: iOS Support for HM-10 Module?

becauseinterwebs, 2017-02-08:
The HM-10 BLE Module is pretty common, but I can't get it to work.  The app cannot discover it.  It works fine in Android and I have installed a few BLE serial test apps on my iPad and they all can see it, however this plugin does not.  Wondering what it would take to add support for the HM-10 for iOS?

becauseinterwebs, 2017-02-08:
Forgot to mention it is a BLE device.

don, 2017-02-08:
I don't have a HM-10. What UUIDs do they use for their SPP service? They'd need to be added to [BLEDefines](https://github.com/don/BluetoothSerial/blob/master/src/ios/BLEDefines.h)

If you want to use HM-10 on iOS and Android, take a look at [cordova-plugin-ble-central](https://github.com/don/cordova-plugin-ble-central). You could use the [bluefruitle example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) as a starting point.

There are tons of HM-10s on Amazon. Do you have a recommendation on a good one to buy or they all the same?

becauseinterwebs, 2017-02-08:
I am using LightBlue on iOS to query the device.  The only service ID I see is 

800A7023-C1B9-12AC-6DAF-3384C7A8F0A2

You have to be pretty careful to get a real HM-10...there are a ton of clones but they will not work.  This is the one I have:

https://www.amazon.com/gp/product/B0140R8DGQ/

Also, it's not working on Android either.  My phone can pair with it, but when I try to connect via the app it just says it can't connect to it.  I will checkout the ble-central plugin thank you!  Can I get the same functionality?  Really needing the serial comunication part.

becauseinterwebs, 2017-02-08:
This might help as well.  http://www.sunfounder.com/wiki/images/f/f8/Bluetooth_4.0_BLE_module_datasheet.pdf

don, 2017-02-08:
It's not working on Android since plugin only support Bluetooth Classic for Android.

With BLE you shouldn't need to pair, the phone can discover the HM-10. 

cordova-plugin-ble-central provides generic Bluetooth LE support. The [bluefruit example](https://github.com/don/cordova-plugin-ble-central/tree/master/examples/bluefruitle) uses the Nordic serial service and provides a serial like API. You should be able adjust the UUIDs and build a serial interface for the HM-10. If you get stuck, create an issue with some code in the other repo.

don, 2017-03-17:
@becauseinterwebs this is tested with HM-10 and iOS. You'll need to install the plugin from github until I release a new version.

commit 6304b0d5 (2017-03-17): HM-10 fixes #288
commit 11fe412c (2017-03-17): Revert "HM-10 fixes #288"

This reverts commit 6304b0d55db8662d3fbc720552596263db0b2db8.
closed 2017-03-17 by don