don/BluetoothSerial #293: Property 'disconnect' does not exist

Kurlrip, 2017-03-07:
I try to do this : 

BluetoothSerial.disconnect()

but typescript say : Property 'disconnect' does not exist on type 'typeof BluetoothSerial'

My package project : 

`  "dependencies": {
    "@angular/common": "2.4.7",
    "@angular/compiler": "2.4.7",
    "@angular/compiler-cli": "2.4.7",
    "@angular/core": "2.4.7",
    "@angular/forms": "2.4.7",
    "@angular/http": "2.4.7",
    "@angular/platform-browser": "2.4.7",
    "@angular/platform-browser-dynamic": "2.4.7",
    "@angular/platform-server": "2.4.7",
    "@ionic/storage": "^1.1.6",
    "@types/pouchdb": "5.4.28",
    "angular2-signaturepad": "^2.2.0",
    "ionic-angular": "2.0.1",
    "ionic-native": "2.5.1",
    "ionicons": "3.0.0",
    "pouchdb": "^6.1.1",
    "rxjs": "5.1.1",
    "zone.js": "0.7.7"
  },
  "devDependencies": {
    "@ionic/app-scripts": "1.1.2",
    "typescript": "2.1.6"
  },
  "cordovaPlugins": [
    "cordova-plugin-whitelist",
    "cordova-plugin-console",
    "cordova-plugin-statusbar",
    "cordova-plugin-device",
    "cordova-plugin-splashscreen",
    "ionic-plugin-keyboard"
  ]`

don, 2017-03-07:
@Kurlrip [bluetoothSerial.disconnect](https://github.com/don/BluetoothSerial#disconnect) is in the bluetoothSerial.js https://github.com/don/BluetoothSerial/blob/master/www/bluetoothSerial.js#L13-L15 

It looks like a problem with the TypeScript bindings. I suggest adding the missing binding to https://github.com/driftyco/ionic-native/blob/master/src/plugins/bluetoothserial.ts and submitting a pull request to @driftyco.

Kurlrip, 2017-03-07:
thk you.
I m begginer in this language, so i don t think that i can send a pull request.
But how can i say to them that method disconnect is missing ??

Else, i tried to modify my bluetoothserial.ts  but don t work.

don, 2017-03-07:
You can submit an issue saying that disconnect is missing from bluetoothserial.ts

Include links to 
https://github.com/don/BluetoothSerial/blob/master/www/bluetoothSerial.js#L13-L15
and
https://github.com/driftyco/ionic-native/blob/master/src/plugins/bluetoothserial.ts

In the mean time if you declare bluetoothSerial, you can call disconnect directly.

    declare var bluetoothSerial: any; 
    // ...
    bluetoothSerial.disconnect();

Kurlrip, 2017-03-07:
That was I just did. 
I ll try to post tomorrow my issue when I'll find where I must post it 
Thk for your quick answer 😊

Kurlrip, 2017-03-08:
i post here :  https://github.com/driftyco/ionic-native/issues/1151

closed 2017-03-08 by Kurlrip