don/BluetoothSerial #294: Windows 10 support

mbluewolf, 2017-03-14:
Are there plans to add windows 10 support.  Can anyone point me in what do i need to do if not?

don, 2017-03-15:
@mbluewolf @Davide-sd sent a pull request #212 for Windows support. I need to merge it but haven't had the opportunity to test. If you could try using his fork and let me know how it works that would be very helpful.

don, 2017-03-17:
@mbluewolf I had some problems with the Windows 10 code. Can you see if you can run https://github.com/don/BluetoothSerial/commit/67ec991e5abf6254ada90bdb2c3cdd831bb53d81?

mbluewolf, 2017-03-17:
Hello, first of all thank you for merging the pull request #212, I download it and have it running on a windows 10 pc with Vs2015, but i needed to modify the chat example since the touchstart event was not firing so I used 'click' instead, Also I had to modify a couple lines in the onDeviceList function as follows:
                //button.dataset = {};
                //button.dataset.deviceId = device.id;
                button.dataset.deviceId = device.address
button.dataset was complaining of being readonly and device.id was empty.

After the changes I could connect an arduino and a RFID scanner.   The arduino project I used only receives data and worked ok, but the scanner sends data to the pc, and there I have some problem.

To test the scanner  I modified the suscribe call at the connect function as follows:
            bluetoothSerial.subscribe("\r", app.onData, app.onError);
when "RD\r"(read command) is sent to the scanner it sends back some data like "H3005FB63AC1F3681EC880468\r"
then "<OK>\r".
The problem is than I have to send 2,3,4 times the read command and the reading is broken in parts most of the time like:
C1F3681EC880468
K>
H3005FB63AC1F3681EC880468
OK>
<OK>
etc.
I am not sure if the problem is only in the reception or in the send too since it shows irregular.

don, 2017-03-18:
I update the windows branch and fixed some stuff in https://github.com/don/BluetoothSerial/commit/7ad83dc78b5cc684e182898dea07c561d5e7757e

Improvements
 * subscribe and subscribeRawData work better
 * receive data handles unicode
 * long lived callbacks use success callback (needed keepCallback: true)
 * return device.id rather than device.address

Known issues
 * error handling still needs works
 * sometimes the connection is really slow
 * device names are incorrect - there's probably another API to get real names
 * device ids work but are clunky - should parse out the MAC address
 * isEnabled returns false if no devices are paired
 * generated manifest is incorrect

mbluewolf, 2017-03-22:
I tested the version you indicated in a  windowsx86 executable and noticed a)device.id comes with info, device.address is empty, b)the reception is ok now!, c)but there is a problem after disconnection at the point where is trying to present the mainPage again, the error shows in the base.js code in the images attached, thank you...
[image]
[image]
[image]

mbluewolf, 2017-03-23:
hmmm, sorry I didn't mean to close the issue...

mezykr, 2017-03-24:
Hi,

When I run bluetoothSerial.connect it returns for me array which contains my device (macbook). As id i see: "\\?\BTHENUM#{00001101-0000-1000-8000-00805f9b34fb}_LOCALMFG&000f#8&260e5458&0&9801A78C5686_C00000000#{b142fc3e-fa4e-460b-8abc-072b628b3c70}"

but when I'm trying to run:

bluetoothSerial.connect("\\?\BTHENUM#{00001101-0000-1000-8000-00805f9b34fb}_LOCALMFG&000f#8&260e5458&0&9801A78C5686_C00000000#{b142fc3e-fa4e-460b-8abc-072b628b3c70}", s, f)

it returns for me:

Failed to connect to server, with error: WinRTError: Element not found.

I'm able to read/write data between my mac and win10 pc using node serialport, so connection should be fine ...

Any ideas what I'm doing wrong?

Thanks

don, 2017-03-24:
@mbluewolf Sounds like there's a bug in the disconnect process. If you find a fix, please send a pull request against the Windows branch.

don, 2017-03-24:
@mezykr unfortunately the name returned by the JavaScript API is horrible. I'm wondering if the "\?\" in the id is some unicode devices that are getting converted incorrectly.

For testing I used this example https://github.com/don/BluetoothSerial/tree/windows/examples/Chat . Can you see if that works with your hardware?

The C# API for Windows Phone used PeerFinder so I was able to get the Name and MAC address of the code https://github.com/don/BluetoothSerial/blob/master/src/wp/BluetoothSerial.cs#L41 and
https://github.com/don/BluetoothSerial/blob/master/src/wp/BluetoothSerial.cs#L328-L341

Maybe there's an equivalent API to get better names and addresses in UWP?

mbluewolf, 2017-03-26:
**@doncoleman** i found the problem, in BluetoothSerial.js the function receiveStringLoop gets executed after disconnect even if unsuscribe was called before disconnect, that generates an error.   Adding a check on the socket solved the error:
var receiveStringLoop = function (reader) {
	if(socket){
		// read one byte at a time
		reader.loadAsync(1).done(function (size) {
........

Tested connecting / disconnecting to more than one device and works fine now, hope is useful for you and can include it a release.

mezykr, 2017-03-28:
Hi, I've found where my issue was located... my id contains "\" signs and when I was using js console to call connect method and paste my id inside it was used as escape character and simple skipped... For now I'm able to connect to my device - thanks for help!

I have also a small question about plugin.xml... I'm not super familiar with this structure but when I'm trying to build app for windows 10 (universal platform) it's working fine but in case when I want to build windows 8.1 I see app manifest validation error. What I did is instead of:

<DeviceCapability Name="bluetooth.rfcomm">
			<Device Id="any">
				<Function Type="name:serialPort" />
			</Device>
		</DeviceCapability>

I've replaced it with:

`<m2:DeviceCapability Name="bluetooth.rfcomm">
			<m2:Device Id="any">
				<m2:Function Type="name:serialPort" />
			</m2:Device>
		</m2:DeviceCapability>`
 
and for now I'm able to build also win 8.1, do you have any idea what does this **m2:** ?

mezykr, 2017-03-29:
I have another question ... is it possible to connect android device with windows 10 device over bluetooth spp ? I was trying to set incoming/outgoing ports in bt manager in system but I was not able to ...

toregua, 2017-05-11:
Hi, 
First thanks for this great plugin and your reactivity
I wanted to know when it was planned to merge the windows  10 branch with the master branch?
We are developing a project under windows 10 with Ionic 3 and we are waiting for this merge.
Thanks in advance

don, 2017-07-07:
@toregua eventually I'd like to get this merged, but my clients don't need this yet and I haven't really tested it.

Have you been using the fork? Is it working well?

Danteam, 2018-05-21:
Hi @don ,
as already done by @toregua i want to Thank you for this plugin and your support.

I need to know if you or someothers already tested and worked with the win universal or know how to add the windows universal support to this plugin ( I already saw the windows 10 branch but it seems not to work, becase the cordova plugin proxy cannot find the bluetooth serial plugin).
P.S. Sorry for my bad english.

don, 2018-05-25:
Hi @Danteam I have not run this on Windows universal lately. I'm pretty sure it can be made to work, I just need to find a day to work on it. If you manage to get stuff working, please comment here and/or send a pull request.

panawe, 2023-03-12:
> **@doncoleman** i found the problem, in BluetoothSerial.js the function receiveStringLoop gets executed after disconnect even if unsuscribe was called before disconnect, that generates an error. [...]

I am facing this same issue. I added the socket check but as results, sometimes the print does not go through.   @mbluewolf Did you make a fix on top of this? Maybe we need to reopen the socket? Not sure why the socket is getting closed.

mbluewolf, 2023-03-14:
Hello,
I have used this in a couple personal projects, and worked ok, I don’t remember the detail but I think it was a mixed version out of the regular because as you mentioned the master had a bug…
I attach the version I am telling about…

Hope it helps Best regards

panawe, 2023-03-14:
@mbluewolf - Did you forget to attach something?

mbluewolf, 2023-03-14:
Sorry I missed…

panawe, 2023-04-16:
@mbluewolf I don't think you ever attached anything. If you're trying attach, it's not working. Please e-mail or put on ftp server and share the link.

mbluewolf, 2023-04-16:
The email does not make sense to me,  so I am guessing with all related, hope this goes throu…

mbluewolf, 2023-04-16:
> @mbluewolf I don't think you ever attached anything. If you're trying attach, it's not working. Please e-mail or put on ftp server and share the link.

@panawe , i reply first time but forgot attach, i replied again, 2 times with today, but your email is not available in github so i fwd the file to don and here...
[cordova-plugin-bluetooth-serial[2305843009214002959].zip](https://github.com/don/BluetoothSerial/files/11242998/cordova-plugin-bluetooth-serial.2305843009214002959.zip)