don/BluetoothSerial #296: Bundled data transfer & slow data rates

japborst, 2017-03-21:
I'm sending data to a BT chip that supports both low-energy and normal, but I use regular BT with pairing and all. In the current set-up I find two issues: 
 - I cannot send more than 120 bytes at a time, without crashing the chip; however it should be able to deal with larger data rates
 - Packages arrive bundled

What happens is that when writing separate sets of bytes, the data arrives bundled. I send 16 bytes at a time using a `U8intArray`. If I send them as a for loop the data is sent too quickly, so I use a timeout after each to avoid the issue. This leads to a timeout of 100ms after each 16 bytes, which is ridiculous and causes extremely slow data transfer.

You have any idea how to circumvent this or better make use of callback functions to wait for the next bytes to sent? Do you know if Bluetooth data transfer is device dependent or should it behave equally amongst all Android devices?

don, 2017-03-24:
You don't say if you're using iOS or Android. I'll assume iOS since Bluetooth Low Energy is really slow and horrible sending large amounts of data.

Crashing the chip with 120bytes sounds like a problem on the chip side. iOS can send up to 500 bytes if the peripheral supports it.

Wait for the success callback from write before sending the next packet. If your hardware supports write (as opposed to writeWithoutResponse) the callback will be called after the write confirmation comes back from the peripheral. If the peripheral supports writeWithoutResponse, you need a delay of at least 30ms between writes.

Bluetooth Low Energy is always going to be impossibly slow for large data transfer, but it can work if you're patient. Android uses Bluetooth classic so it should be fast.

japborst, 2017-03-24:
Yes so here it becomes interesting. On iOS it is actually faster. We use the write over LE without any delay and it works beautifully. On Android we actually have to use the 100ms delay with the regular write function that executes on the success callback for the data transfer to work. Over classic, you'd assume that much higher data transfers would be possible. Chip should support at least BT 3

don, 2017-03-24:
@japborst you *should* see much faster data transfer on Android. If I recall correctly, it should work with Bluetooth 2.1+.

Do you see this behavior on all Android devices or just some?

japborst, 2017-03-24:
The same behaviour on the Android devices we have laying around. Will test on others. Would this be Android version dependent or BT chip of the Android device?

japborst, 2017-03-24:
When sending the data from a Windows PC using a python script, we don't need any delay as long as we're below 120bytes. So this definitely seems Android related

don, 2017-03-24:
Bluetooth Classic is pretty stable on Android, but in my experience there are always some Android versions and devices were things are broken. It could be dependent on the Bluetooth chipset. 

It's possible there is some configuration that Android or your hardware needs to work together nicely. Tough to say for sure.

closed 2017-08-18 by don