don/BluetoothSerial #297: iOS stopped working

bucktaylor, 2017-03-26:
HI we've been using this plugin for a while, and it was working in our app for both iOS and Android fine up to our last compilation and release in November 2016.

We recently re-compiled a new version of our app with no code changes to our bluetooth implementation, and now both our internal and external testers are getting no connection on iOS devices.

For older devices without LE support - we get an instant "Bluetooth is not enabled", and for those with LE, we get "Did not find any BLE peripherals.

Newer Android devices seem to connect without issue (haven't tried any older ones yet).

Any chance you can help us out with some ideas on whats happening here?

don, 2017-03-31:
v0.4.6 added Android permissions. For iOS 0.4.5 and 0.4.6 should be the same.

Maybe you need to upgrade Cordova? Maybe it's something with the newer versions of Xcode and iOS? Some early builds of iOS 10 had problems with some Bluetooth peripherals that worked OK with iOS 9. I don't really have enough info to diagnose.

bucktaylor, 2017-03-31:
Hi Don,

Thanks – we found out yesterday that in the previous version of your plugin one of our previous developers had modified your library to work with the nBlue - BlueRadios chipset and never documented it! Whenever we ran the CLI it was brining in the new plugin – not the modified one. Sorry for the trouble.

Buck

don, 2017-03-31:
Create a pull request if you'd like nBlue radios to be included in future releases.

closed 2017-03-31 by don