don/BluetoothSerial #302: BluetoothSerial doesn't work in Android

catarina2, 2017-04-26:
Hello, 

I would like some help please on this issue:
I have installed phoneGap, and I'm following one of the example of the bluetooth serial, to interact android with hardwin by bluetooth hc-5. 
This is a part of the code:
bluetoothSerial.connect("20:16:05:16:65:20", app.onconnect, app.ondisconnect); 
And when I'm on the pc it goes to app.onconnect, and on Android it doesn't work! On the pc it shows div.setstatus and on the android it doesn't. I've tried to list the devices on the pc I can, but on Android I can't. I think it has to do with the library bluetooth serial. Do I have to install anything on the Android regarding this library? Or should a place something on the config.xml? I have the library installed and it works on the pc, but on the android it doesn't! How can I solve this?
Thanks from Portugal

bottasergio, 2017-07-12:
@catarina2 Did you solve this? I have the same problem!

don, 2017-07-12:
I recommend pairing the Bluetooth device with your Android device before running your app. 

With Android `bluetoothSerial.list` lists the paired Bluetooth Classic devices.

bottasergio, 2017-07-12:
I have tried this:

`bluetoothSerial.connect('00:??:??:??:??:??', BTConectado , function(){alert('error!');} );`

... and it calls **BTConectado**  so I assume my app is connected to the printer...

This is the function...

`function BTConectado(m) {`
`bluetoothSerial.write('data', function(){alert('OK!');}, function(){alert('ERROR!');});`
`}	`

... it shows **OK!** but nothing happens on printer.

don, 2017-07-12:
@bottasergio Yes, it sounds like you are connected and sending data to the printer. The OK means that the data went to the printer. This plugins is providing a way to deliver data over Bluetooth. You'll need to figure out if your printer what type of data your printer is expecting. It might be text, Printer Command Language (PCL), or some magic control codes.

bottasergio, 2017-07-14:
Great! Solved! I am using a Citizen CMP-30 with CPCL Emulation. On var `data` you must send CPCL commands.

CPCL commands here:
[http://www.citizen-systems.co.jp/support/download/printer/cmp20_30data/E/CMP-30CPCL-CMD002E.pdf](http://www.citizen-systems.co.jp/support/download/printer/cmp20_30data/E/CMP-30CPCL-CMD002E.pdf)

Code:
`var CHAR_ESC = 0x1b;
var LINE_FEED = 0x0A;
var CARRIAGE_RETURN = 0x0D;
data = "! 0 200 200 500 1" + CARRIAGE_RETURN + LINE_FEED + '\n' +
						   "TEXT 4 0 0 45 TEST" + CARRIAGE_RETURN + LINE_FEED + '\n' +
							"TEXT 4 0 0 95 Text Text" + CARRIAGE_RETURN + LINE_FEED + '\n' +
							"INVERSE-LINE 0 45 145 45 45" + CARRIAGE_RETURN + LINE_FEED + '\n' +
                    "PRINT" + CARRIAGE_RETURN + LINE_FEED + '\n' 
`

Thanks @don for your help!

don, 2017-07-14:
Glad it's working. Thanks for the update.

closed 2017-07-14 by don