don/BluetoothSerial #308: Android: bluetoothSerial.connect does not callbacks

kaddyadriano, 2017-06-06:
``bluetoothSerial.connect`` does not fire the success or error callbacks at first. When I try again it fires the error callback with 'Unable to connect to device'. I have tried different devices but the problem still the same. I am thinking the success callback is not being fired and when I try to connect again I get an error because it's already connected I am not sure. Can you please help!

don, 2017-07-07:
Do you have some code I can see that duplicates this?

Also, make sure you wait for deviceready to be fired before calling any bluetoothSerial functions.

AsreshSharma, 2017-08-09:
same issue.
bluetoothSerial.connect does not fire the success or error callbacks at first.

import { Component } from '@angular/core';
import { AlertController } from 'ionic-angular';

import { BluetoothSerial } from '@ionic-native/bluetooth-serial';

import { NavController } from 'ionic-angular';

//import { Observable } from 'rxjs';

import { WelcomePage } from '../welcome/welcome';

@Component({
  selector: 'page-home',
  templateUrl: 'home.html'
})

export class HomePage {

  unpairedDevices: any;  
  unpairedDevicesError:any;
  pairedDevices: any;
  gettingDevices: Boolean;

  btNearBy: boolean;

  constructor(private bluetoothSerial: BluetoothSerial, private alertCtrl: AlertController, public navCtrl: NavController)
  {
  	bluetoothSerial.enable();    
    this.unpairedDevices = null;
    this.startScanning(); 
  }
  
  startScanning() 
  {
    this.bluetoothSerial.enable();
    this.pairedDevices = null;
    this.unpairedDevices = null;
    this.unpairedDevicesError=null;
    this.gettingDevices = true;
    
    this.bluetoothSerial.discoverUnpaired().then((success) => { 
              
              var arrayWithDuplicates = success;

              var uniqueArray =this.removeDuplicates(arrayWithDuplicates,"class");

              console.log("uniqueArray is: " + JSON.stringify(uniqueArray));

              console.log(uniqueArray);

              this.unpairedDevices=uniqueArray;
              this.gettingDevices = false;
    },    
    (err) => {
      console.log(err);
      var errorvalue=['Problem while discovering'];
      this.unpairedDevicesError=errorvalue;
    })

    this.bluetoothSerial.list().then((success) => {
      this.pairedDevices = success;
    },
    (err) => {
     console.log(err);
    })
  }

  removeDuplicates(originalArray, prop) 
  {
     var newArray = [];
     var lookupObject  = {};

     for(var i in originalArray) 
     {
        lookupObject[originalArray[i][prop]] = originalArray[i];
     }

     for(i in lookupObject) 
     {
         newArray.push(lookupObject[i]);
     }
      return newArray;
  }

  //success = (data) => console.log('success data : ',data);

  //fail = (error) => console.log('Failure Data : ',error);

  success(data)
  {
      console.log('success data : '+ JSON.stringify(data));
      this.bluetoothSerial.available().then(data =>{

        console.log("Available " + JSON.stringify(data));

        this.bluetoothSerial.read().then(data =>{

          console.log("Read " + JSON.stringify(data));
          this.navCtrl.push(WelcomePage);

        });
      });
  }

  fail(error)
  {
    console.log('Failure Data : '+ JSON.stringify(error));
  }

  selectDevice(address: any)
  {              
    let alert = this.alertCtrl.create({
      title: 'Connect',
      message: 'Do you want to connect with?',
      buttons: 
      [
            {
              text: 'Cancel',
              role: 'cancel',
              handler: () => {
              console.log('Cancel clicked');
            }
      },
            {
              text: 'Connect',
              handler: () => 
              {
                this.bluetoothSerial.connect(address).subscribe(this.success, this.fail);  
              }
            }
      ]
    });
    alert.present();

  }

  disconnect() {
    let alert = this.alertCtrl.create({
      title: 'Disconnect?',
      message: 'Do you want to Disconnect?',
      buttons: [
        {
          text: 'Cancel',
          role: 'cancel',
          handler: () => {
            console.log('Cancel clicked');
          }
        },
        {
          text: 'Disconnect',
          handler: () => {
            this.bluetoothSerial.disconnect();
          }
        }
      ]
    });
    alert.present();
  }

  searchNearBy()
  {    
    
    let chkbox=this.btNearBy;
    if(chkbox != true)
    { 
      this.gettingDevices=false;
    }
    else
    {      
      this.startScanning();
    }

  }

}

don, 2017-08-18:
I can not duplicate this. I'm getting callback when the connection succeeds using a good MAC address and when the connection fails using an invalid MAC address.

Please see if this sample project works with your hardware https://github.com/don/bluetooth-serial-issue-308

Tested on 
Cordova 7.0.1
BluetoothSerial 0.4.7
Nexus 5X running Android 7.1.2
macOS 10.12.6 
Nodejs 6.11.2

closed 2017-09-13 by don

hossein-hbb, 2018-09-07:
Hi everyone.
I have very simple question.
I’m using bluetooth-serial for connect and receive from arduino and i use readUntil(’\n’) for receive from arduino
I use readUntil with setInterval and inside constructor but I think this way wrong because it’s has very lag .
I need help for this situation.
`
constructor(private bluetoothSerial: BluetoothSerial, public navCtrl: NavController, public navParams: NavParams) {
    setInterval(() => {
      this.bluetoothSerial.readUntil('\n').then((data: any) => {
        if (data.indexOf("code:") != -1) {
          var si = data.indexOf("code:");
          this.code = data.substring((si + 5), data.length);
        }
      });
    }, 0);
  }`
I need read from arduino and find some information from sender device and i use substring for looking for some data and my situation very lag…
Thanks for help