don/BluetoothSerial #309: ERROR: Method 'discoverUnpaired:' not defined in Plugin 'BluetoothSerial'

dtaalbers, 2017-06-16:
I've installed version`<plugin name="cordova-plugin-bluetooth-serial" spec="~0.4.7" />` of the plugin in my ionic (v2) app. Androids works fine, but the same code does not work on my IOS devices. I am trying to connect to this [Bluetooth Barcode Scanner](https://www.thr.nl/thr-bluetooth-barcode-scanner-0115-10.html). 

Like I said android works fine. I can use the following functions just fine:

```
isEnabled()
isConnected()
list()
connect()
subscribeRawData
discoverUnpaired()
showBluetoothSettings()
```

But I am getting errors when using some of the functions on the IOS device. See stacktrace that I am getting from XCODE DEBUG from the `discoverUnpaired` function. I do a `list()` before the `discoverUnpaired` function, but it does not find any devices even though I **know** some are available.

```
2017-06-16 16:28:00.834179+0200 MyApp[19356:5721849] Scanning for BLE Peripherals
2017-06-16 16:28:00.835687+0200 MyApp[19356:5721849] scanForPeripheralsWithServices
2017-06-16 16:28:03.835733+0200 MyApp[19356:5721849] Stopped Scanning
2017-06-16 16:28:03.835913+0200 MyApp[19356:5721849] Known peripherals : 0
2017-06-16 16:28:03.835978+0200 MyApp[19356:5721849] List of currently known peripherals :
2017-06-16 16:28:03.903981+0200 MyApp[19356:5721849] ERROR: Method 'discoverUnpaired:' not defined in Plugin 'BluetoothSerial'
2017-06-16 16:28:03.904116+0200 MyApp[19356:5721849] -[CDVCommandQueue executePending] [Line 142] FAILED pluginJSON = ["BluetoothSerial832227986","BluetoothSerial","discoverUnpaired",[]]
```

The function `showBluetoothSettings`
```
2017-06-16 16:30:40.770141+0200 MyApp[19356:5721849] ERROR: Method 'showBluetoothSettings:' not defined in Plugin 'BluetoothSerial'
2017-06-16 16:30:40.770246+0200 MyApp[19356:5721849] -[CDVCommandQueue executePending] [Line 142] FAILED pluginJSON = ["BluetoothSerial832227987","BluetoothSerial","showBluetoothSettings",[]]
```

This is a bit weird because the `isEnabled()` function works fine also on the IOS device. I am probably doing something wrong. Would appreciate some help!

Device info:
```
IOS:
Iphone 6s+  (does not work)
Version: 10.3.2

ANDROID:
Samsung S7 Edge
Version 7.0
```

Typescript file:
```
import { Component } from '@angular/core';
import { BluetoothSerial } from '@ionic-native/bluetooth-serial';

@Component({
    templateUrl: 'bluetooth-scanner-.html'
})
export class BlueToothScannerPage {

    /**
     * The available paired bluetooth devices
     */
    public paired_devices: Array<any> = [];;
    /**
     * The available unpaired bluetooth devices
     */
    public unpaired_devices: Array<any> = [];
    /**
     * A flag that indicates whether we are scanning for bluetooth devices or not
     */
    public is_scanning: boolean = false;

    constructor(
        private bluetooth_controller: BluetoothSerial
    ) { }

    /**
     * Checks if bluetooth is enabled or not
     */
    public async is_enabled(): Promise<void> {
        let is_bluetooth_enabled;
        try {
            await this.bluetooth_controller.isEnabled();
            is_bluetooth_enabled = true;
        } catch (e) {
            is_bluetooth_enabled = false;
        }
        if (!is_bluetooth_enabled) {
            // Not enabled
        }
    }

    /**
     * Scans for available bluetooth devices
     */
    public async scan(): Promise<void> {
        try {
            this.paired_devices = [];
            this.unpaired_devices = [];
            this.is_scanning = true;
            this.paired_devices = (await this.bluetooth_controller.list());
            this.unpaired_devices = (await this.bluetooth_controller.discoverUnpaired())
                // Filter out the devices that have no name
                .filter(x => x.name)
                // Filter out devices that are already shown in the paired devices list
                .filter(x => !this.paired_devices.some(y => y.name == x.name));
            this.is_scanning = false;
        } catch (e) {
            this.is_scanning = false;
            this.notification_service.notify_error(e, '[bluetooth/scan];');
        }
    }

    /**
     * Opens the bluetooth settings
     */
    public open_settings(): void {
        this.bluetooth_controller.showBluetoothSettings();
    }
}

```

dtaalbers, 2017-06-19:
@don Sorry to tag you in here. But do you have any idea what could cause this issue? The shipment on the app I'm is delayed on this issue I'm having.  Its works beautifully on android, but can't get the same code to work with IOS. A response would be nice, so I can decide whether to wait for a solution or try something else (what, I don't know)

don, 2017-07-07:
@dtaalbers My guess is the barcode scanner is Bluetooth Classic since things work on Android.

This plugin uses Bluetooth Low Energy for iOS, which is probably why iOS is failing for you.

Some people have forked this plugin to use the External Accessories Framework to access Bluetooth Classic on iOS. That requires an Apple MFi license. Another solution can be to get the iOS SDK from the barcode scanner manufacturer and create a new plugin that uses the SDK. (Sometimes this requires a MFi license too.)

closed 2017-07-22 by don