don/BluetoothSerial #311: Getting error info from connect atempt

vitorgomateus, 2017-06-18:
Maybe not the proper place, but I've been all over trying to solved this, tried going through the source, but I'm helpless and I really need help.

I need to find out if a paired device is in range without having to successfully connect.
I believe I can find that out from the resulting error of a failed connect attempt.
I tried to receive the argument in the connectFailure callback, to no success.
Don't know if the plugin does not have this implemented or I'm just doing it wrong.

Thank you!
`
ondisconnect: function(something) {`
`        errorDiv.innerHTML = Json.stringify(something);`
 `       errorDiv.innerHTML += "<br/>error might be: "+something;`
`}`

don, 2017-07-07:
When a connection fails, the failureCallback of [bluetoothSerial.connect](https://github.com/don/BluetoothSerial#connect) will be called. Note that the failure callback it a long running callback. It may be called immediately or it can be called much later when the connection disconnects.

As for finding if a device is in range, on iOS (BLE) you'd do a scan to see if the device was in range. with Android (Bluetooth Classic) there are probably some APIs but I'm not sure they're exposed. For iOS try [bluetoothSerial.list](https://github.com/don/BluetoothSerial#list) to see a list of devices near you.

vitorgomateus, 2017-07-09:
It turns out I had a misunderstanding of list and listUnpaired. ListUnpaired did the trick. Still a detailed connection error might have been more precise, but it was quite successful. I had my master's delivery this friday, but thank you! Your plugin really made it all much easier! and compared with another more complex plugin  the simplicity was much appreciated. Great job Don!

closed 2017-07-09 by vitorgomateus