don/BluetoothSerial #314: Disconnect function not available

mohammed-mudassir, 2017-07-06:
Property  disconnect does not exist on type 'Typeof BlutoothSerial'

mohammed-mudassir, 2017-07-06:
@don please reply on this issue

don, 2017-07-07:
This is a duplicate of #293. The TypeScript bindings are part of [Ionic native](https://ionicframework.com/docs/native/) not this plugin.

It looks like `disconnect` was added to the api https://ionicframework.com/docs/native/bluetooth-serial/#disconnect in https://github.com/ionic-team/ionic-native/issues/1151.

Maybe you have an older version of Ionic? I just tested with Ionic 3.4.0 and it looks OK.

Try this

    ionic start disconnect-test blank
    cd disconnect-test
    ionic cordova plugin add cordova-plugin-ble-central
    npm install --save @ionic-native/ble

Open src/pages/home/home.ts in your editor and import BLE, add to the constructor, then call disconnect

	import { Component } from '@angular/core';
	import { NavController } from 'ionic-angular';
	import { BLE } from '@ionic-native/ble';

	@Component({
	  selector: 'page-home',
	  templateUrl: 'home.html'
	})
	export class HomePage {

	  constructor(public navCtrl: NavController, private ble: BLE) {
	    // Just testing to make sure ble.disconnect works
	    ble.disconnect('bogusid');
	  }

	}

Ensure everything compiles without errors

    ionic cordova build android

mohammed-mudassir, 2017-07-10:
Hey @don thanks for reply my issue is fixed now

closed 2017-07-12 by don