don/BluetoothSerial #315: Bluetooth auto pairing

mohammed-mudassir, 2017-07-10:
@don 

Can you help me in the following case?
I am developing one App wherein I am trying to connect Bluetooth automatically without user need to put passcode. 
I have gone through plugin files but exactly where shall I need to do changes.

The case will be like: 
I will have opponent MAC/UUID to connect to the devices. So using this user can connect to the Bluetooth in the background without taking user input

mumairofficial, 2017-07-11:
Actually functionality you want is exist here #112 @don didn't merge that branch because author used reflection. Unfortunately since it's old, there are now conflicts.

If you want this functionality you can use the branch for the pull request from the authors github https://github.com/stakutis/BluetoothSerial/tree/stakutis

#### Backup your project
Remove the official version of the plugin
```cordova plugin rm cordova-plugin-bluetoothserial```

Try the alternate version with the feature you want
`cordova plugin add https://github.com/stakutis/BluetoothSerial/tree/stakutis`

don, 2017-07-12:
@mumairofficial Thanks for the detailed answer.

closed 2017-07-12 by don