don/BluetoothSerial #317: bluetoothSerial.Read how to return barcodes separately

iakwvina, 2017-07-13:
Hi,
When I am scanning multiple codes, the bluetoothSerial.Read returns all of them in one line. Their length isn't the same. Is there any code to  take each code length to separate them later?

don, 2017-07-13:
If the data is delimited with something like a line feed you could use [bluetoothSerial.readUntil](https://github.com/don/BluetoothSerial#readuntil).

iakwvina, 2017-07-17:
I noticed that the data had a empty character, which was not visible inspecting the data element, but only using data.length(). So I did something like this, to separate them:  
```
var correctData = data.split("");
var correctData_arr = correctData.join("").trim().split(/\s/); 
```

don, 2017-07-22:
Thanks for the update.

closed 2017-07-22 by don