don/BluetoothSerial #318: Can't read data from odb2

vladimircojuhari, 2017-07-19:
hello, i have a problem. when im trying to read data from a bluetooth divice(odb2 in this example) i receive just empty string.
this is my code: 
 this.bluetoothSerial.read().then(result => {
             var data = new Uint8Array(result);
             console.log(String.fromCharCode.apply(null, data));
         })

this.bluetoothSerial.subscribeRawData().subscribe(items => console.log(items));
---------------------------
 OnInit(){
         this.bluetoothSerial.subscribeRawData().subscribe(items => console.log(items));
}
subscireraw data doesnt trigger at all. pls help me

don, 2017-07-22:
Assuming you're using Android, the plugin will get you a Bluetooth Classic connection to your OBDII connector. You're going to need to figure out what commands to send to make it send data back to you. This will be specific to the Bluetooth hardware you're using. Many of those OBDII devices use EML327 clones. Hopefully you can find some documentation on how the communication works. I haven't written code for those devices.

closed 2017-07-22 by don

kherP, 2017-09-11:
Hi, are the commands sent using bluetooSerial.write()?