don/BluetoothSerial #324: Empty unpaired devices list on iOS10

lmark83, 2017-08-03:
Hi,

I'm using the plugin in order to connect my application to a Bluetooth Thermal Printer; it work fine on Android but on iOS 10 when I call the list function I always get 0 devices.

Am I missing any configuration or iOS 10 is not supported by the plugin?

Thank you in advance.
Mark

don, 2017-08-18:
iOS only supports some Bluetooth Low Energy devices with SPP like services. Currently RedBearLab BLE hardware, Adafruit Bluefruit LE, Laird BL600, and BlueGiga are supported. If your printer has as a BLE service, see https://github.com/don/BluetoothSerial#supporting-other-ble-hardware

closed 2017-08-18 by don

lmark83, 2017-08-18:
Thank you Don! 

Could you please suggest me any vendor that produces thermal printers equiped with the BLE modules you listed? Unfortunately at the moment I am not abke to edit the defines but I still want to keep your plugin in my app.

Thank you again
Mark

don, 2017-08-18:
Sorry, I haven't done much with thermal printers.