don/BluetoothSerial #327: How To connect 2 spp device at the same time

tiaraxin, 2017-08-16:
Excuse me, 
How can I connect 2 spp devices simultaneously.

var btSerial1 = cordova.require("cordova-plugin-bluetooth-serial.bluetoothSerial");
var btSerial2 = cordova.require("cordova-plugin-bluetooth-serial.bluetoothSerial");

When using one of them is working well.
When I connect both of them, exceptions catched:
--------------------------------------------------------------------------------------------
08-16 11:17:44.045 29133-29133/com.example.hello I/chromium: [INFO:CONSOLE(159)] "connecting 20:15:05:26:53:69", source: file:///android_asset/www/plugins/cn.net.kage.cordova/www/ExtraInfo.js (159)
08-16 11:17:44.045 29133-30917/com.example.hello I/BluetoothSerialService: BEGIN mConnectThread SocketType:Secure
08-16 11:17:44.045 29133-29193/com.example.hello I/BluetoothSerial: MESSAGE_STATE_CHANGE: 2
08-16 11:17:44.045 29133-29193/com.example.hello I/BluetoothSerial: BluetoothSerialService.STATE_CONNECTING
08-16 11:17:44.050 29133-29133/com.example.hello I/chromium: [INFO:CONSOLE(159)] "connecting 74:F0:7D:E0:0F:98", source: file:///android_asset/www/plugins/cn.net.kage.cordova/www/ExtraInfo.js (159)
08-16 11:17:44.050 29133-30917/com.example.hello I/BluetoothSerialService: Connecting to socket...
08-16 11:17:44.050 29133-30917/com.example.hello E/BluetoothSerialService: java.io.IOException: socket closed
08-16 11:17:44.050 29133-30917/com.example.hello I/BluetoothSerialService: Trying fallback...
                                                                           
                                                                           [ 08-16 11:17:44.050 11486:11502 E/         ]
                                                                           ### ASSERT : external/bluetooth/bluedroid/main/../btif/src/btif_dm.c line 1176 unhandled search services event (6) ###
08-16 11:17:44.050 29133-30918/com.example.hello I/BluetoothSerialService: BEGIN mConnectThread SocketType:Secure
08-16 11:17:44.050 29133-30917/com.example.hello W/BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
08-16 11:17:44.055 29133-29193/com.example.hello I/BluetoothSerial: MESSAGE_STATE_CHANGE: 2
08-16 11:17:44.055 29133-29193/com.example.hello I/BluetoothSerial: BluetoothSerialService.STATE_CONNECTING
                                                                    
                                                                    [ 08-16 11:17:44.055 11486:11502 E/         ]
                                                                    ### ASSERT : external/bluetooth/bluedroid/main/../btif/src/btif_dm.c line 1176 unhandled search services event (6) ###
08-16 11:17:44.055 29133-30918/com.example.hello I/BluetoothSerialService: Connecting to socket...
08-16 11:17:44.055 29133-30918/com.example.hello W/BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback
08-16 11:17:44.865 29133-30917/com.example.hello I/BluetoothSerialService: Connected
08-16 11:17:44.865 29133-30928/com.example.hello I/BluetoothSerialService: BEGIN mConnectedThread
08-16 11:17:44.875 29133-29193/com.example.hello I/BluetoothSerial: irxon
08-16 11:17:44.875 29133-29193/com.example.hello I/BluetoothSerial: MESSAGE_STATE_CHANGE: 3
08-16 11:17:44.875 29133-29193/com.example.hello I/BluetoothSerial: BluetoothSerialService.STATE_CONNECTED
08-16 11:17:45.060 29133-30918/com.example.hello I/BluetoothSerialService: Connected
08-16 11:17:45.060 29133-30937/com.example.hello I/BluetoothSerialService: BEGIN mConnectedThread
08-16 11:17:45.060 29133-30928/com.example.hello E/BluetoothSerialService: disconnected
                                                                           java.io.IOException: bt socket closed, read return: -1
                                                                               at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java)
                                                                               at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java)
                                                                               at java.io.InputStream.read(InputStream.java)
                                                                               at com.megster.cordova.BluetoothSerialService$ConnectedThread.run(BluetoothSerialService.java:451)
08-16 11:17:45.075 29133-29133/com.example.hello I/chromium: [INFO:CONSOLE(174)] "attempting to disconnect", source: file:///android_asset/www/plugins/cn.net.kage.cordova/www/ExtraInfo.js (174)
08-16 11:17:45.075 29133-29193/com.example.hello I/BluetoothSerial: SPP-R200II
08-16 11:17:45.075 29133-29193/com.example.hello I/BluetoothSerial: MESSAGE_STATE_CHANGE: 3
08-16 11:17:45.075 29133-29193/com.example.hello I/BluetoothSerial: BluetoothSerialService.STATE_CONNECTED
08-16 11:17:45.075 29133-30937/com.example.hello E/BluetoothSerialService: disconnected
                                                                           java.io.IOException: bt socket closed, read return: -1
                                                                               at android.bluetooth.BluetoothSocket.read(BluetoothSocket.java)
                                                                               at android.bluetooth.BluetoothInputStream.read(BluetoothInputStream.java)
                                                                               at java.io.InputStream.read(InputStream.java)
                                                                               at com.megster.cordova.BluetoothSerialService$ConnectedThread.run(BluetoothSerialService.java:451)
08-16 11:17:45.080 29133-29193/com.example.hello I/BluetoothSerial: MESSAGE_STATE_CHANGE: 0
08-16 11:17:45.080 29133-29193/com.example.hello I/BluetoothSerial: BluetoothSerialService.STATE_NONE
08-16 11:17:45.080 29133-29193/com.example.hello I/BluetoothSerial: MESSAGE_STATE_CHANGE: 0
08-16 11:17:45.080 29133-29193/com.example.hello I/BluetoothSerial: BluetoothSerialService.STATE_NONE
08-16 11:17:45.085 29133-29193/com.example.hello I/BluetoothSerial: MESSAGE_STATE_CHANGE: 0
08-16 11:17:45.085 29133-29193/com.example.hello I/BluetoothSerial: BluetoothSerialService.STATE_NONE
08-16 11:17:45.085 29133-29193/com.example.hello I/BluetoothSerial: MESSAGE_STATE_CHANGE: 0
08-16 11:17:45.085 29133-29193/com.example.hello I/BluetoothSerial: BluetoothSerialService.STATE_NONE
08-16 11:17:45.090 29133-29193/com.example.hello I/BluetoothSerial: MESSAGE_STATE_CHANGE: 0
08-16 11:17:45.090 29133-29193/com.example.hello I/BluetoothSerial: BluetoothSerialService.STATE_NONE
--------------------------------------------------------------------------------------------

don, 2017-08-18:
@tiaraxin multiple connections aren't supported by the plugin. Some people have been able to make this work by customizing the plugin see https://github.com/don/BluetoothSerial/issues/58#issuecomment-171879745

Closed as duplicate of #58

closed 2017-08-18 by don

don, 2017-08-18:
You might want to look at #205 too