don/BluetoothSerial #329: Arduino serial communication

unseen1980, 2017-08-18:
Hi all,

I connected successfully my ionic app to an arduino attached bluetooth module but when I try to write to something to serial the characters are garbage.

[image]
[image]

I tried different speeds etc but no luck. Any ideas?

don, 2017-08-18:
On my Arduino hardware I usually read Bluetooth data from SoftwareSerial and write to Serial. Some radios need [a series of setup commands](https://github.com/don/BluetoothSerial/blob/master/examples/Chat/Arduino/Chat_SeeedShield/Chat_SeeedShield.ino#L57-L69) which I usually copy and paste from an example or the documentation. Unfortunately I haven't used the HC-06. Some my other sample Arduino code might give you some ideas to try https://github.com/don/BluetoothSerial/tree/master/examples/Chat/Arduino

unseen1980, 2017-08-18:
Thanks, don, I'll give it a try and get back to you!

closed 2017-09-13 by don