don/BluetoothSerial #330: NullPointer Exception Crash

GPlay97, 2017-08-25:
I'm using this plugin on my app.

I've received several crash reports on the play developer console with the following messages:

> java.lang.NullPointerException: 
>   at com.megster.cordova.BluetoothSerialService$ConnectThread.run (BluetoothSerialService.java:373)

Is this a known issue?

don, 2017-09-13:
I haven't seen that. Any idea how to reproduce it?

GPlay97, 2017-09-17:
Unfortunately not. My app also ensures that you can not connect to an invalid device. But maybe the device got unpaired over the android settings so the connection fails? But I really don't have any idea to reproduce it. Only information I have is that it occurs on the specified line in the bluetooth serial service file.

GPlay97, 2017-10-13:
To give you an update:

The reported nullPointer Exception:

> java.lang.NullPointerException: 
  at com.megster.cordova.BluetoothSerialService$ConnectThread.run (BluetoothSerialService.java:373)

is still happening sometimes. I can see this on the Google Play Developer Console.
It happened on Android 6 and Android 7.1 devices. During this week, it occurred 3 times.

Furthermore, I've received this NullPointerException today, which also occurs with your plugin:

> java.lang.NullPointerException: 
  at com.megster.cordova.BluetoothSerialService$ConnectThread.cancel (BluetoothSerialService.java:408)
  at com.megster.cordova.BluetoothSerialService.start (BluetoothSerialService.java:94)
  at com.megster.cordova.BluetoothSerialService.connectionFailed (BluetoothSerialService.java:232)
  at com.megster.cordova.BluetoothSerialService.access$500 (BluetoothSerialService.java:27)
  at com.megster.cordova.BluetoothSerialService$ConnectThread.run (BluetoothSerialService.java:392)

don, 2017-10-23:
Does the play developer console give you any details about which phones this happens on?

The best bet would be to try and duplicate this in your app and create something I can reproduce.

GPlay97, 2017-10-24:
It does.
I'll try my best to try to reproduce it on my own to give you more details.

Device | Occurences | Percentage
-- | -- | --
serranoltexx | 3 | 37,5 %
Moto G (3rd Gen) (osprey_u2) | 2 | 25,0 %
PRIV by BlackBerry (venice) | 1 | 12,5 %
LG G4 (p1) | 1 | 12,5 %
OnePlus5 (OnePlus5) | 1 | 12,5 %

Android Version | Occurences | Percentage
-- | -- | --
Android 7.1 | 4 | 50,0 %
Android 6.0 | 4 | 50,0 %

GPlay97, 2018-08-02:
The problem still exists - a lot of time. Even when using this very simple, it occurs randomly sometimes.

``
java.lang.NullPointerException: 
  at com.megster.cordova.BluetoothSerialService$ConnectThread.cancel (BluetoothSerialService.java:408)
  at com.megster.cordova.BluetoothSerialService.start (BluetoothSerialService.java:94)
  at com.megster.cordova.BluetoothSerialService.connectionFailed (BluetoothSerialService.java:232)
  at com.megster.cordova.BluetoothSerialService.access$500 (BluetoothSerialService.java:27)
  at com.megster.cordova.BluetoothSerialService$ConnectThread.run (BluetoothSerialService.java:392)
``

Furthmore new issues randomly popped out recently:

``
java.lang.StringIndexOutOfBoundsException: 
  at java.lang.StringFactory.newStringFromBytes (StringFactory.java:71)
  at java.lang.StringFactory.newStringFromBytes (StringFactory.java:54)
  at com.megster.cordova.BluetoothSerialService$ConnectedThread.run (BluetoothSerialService.java:452)
``

``
java.lang.NullPointerException: 
  at com.megster.cordova.BluetoothSerialService$ConnectThread.cancel (BluetoothSerialService.java:408)
  at com.megster.cordova.BluetoothSerialService.start (BluetoothSerialService.java:94)
  at com.megster.cordova.BluetoothSerialService.connectionFailed (BluetoothSerialService.java:232)
  at com.megster.cordova.BluetoothSerialService.access$500 (BluetoothSerialService.java:27)
  at com.megster.cordova.BluetoothSerialService$ConnectThread.run (BluetoothSerialService.java:392)
``

The first one happens very often (but random - don't know when) - but it happens on all android versions. And on different phones.

Wulfex, 2018-10-30:
@don The easiest way I've found to reproduce this issue in my own app is to connect a bluetooth device to the phone app, then turn off bluetooth via the phone settings. Not sure the best way to handle this case. I guess ideally in our scenario we would use the enable function to prompt the user to enable bluetooth again, but I think the app crashes before being able to do anything.

GPlay97, 2018-10-31:
@Wulfex Thats it! Wow, nice! Thank you!