don/BluetoothSerial #333: Unable to connect BL600 VSP mode

jordanChalet, 2017-09-15:
Hi all,

I'm trying to connect to BL600(VSP hardware mode). My BluetoothSerial.connect function works  with Iphone 5c (BL600 paired) and failed with Nexus5 :  connect() function return "unable to connect to device".
I added .list() and .discoveredUnpaired() but nothing change. I use MAC address for Android and UUID for IOS.

Thanks for your help !

```var bleConnection = function (id){
        bluetoothSerial.connect(
            id,
            function(res){ 
                console.log("Bluetooth is connected");
                bluetoothSerial.write(
                    stringToBytes("1"),
                    function(res){
                        $scope.progressval = 30;
                        $scope.isConnected = 1;
                        bluetoothSerial.subscribe('\n', receivedData, onError);
                        checkConnexion = $interval( function() {
                                bluetoothSerial.isConnected(
                                    function() {
                                        console.log("Bluetooth is connected");
                                        $scope.isConnected = 1;
                                    },
                                    function() {
                                        console.log("Bluetooth is not connected");
                                        $scope.isConnected = 0;
                                        $scope.data = [];
                                        $interval.cancel(checkConnexion);
                                    }
                                );
                            }, 1000); 
                    }, 
                    function(err){
                        $interval.cancel($scope.stopinterval);
                    }
                );
            },function(err){
                console.log(err);
                $interval.cancel($scope.stopinterval);
                $scope.isConnected = 0;
                $scope.data = [];
                $scope.progressval = 0;
            }
        );
    }
    $scope.scan = function(){
        $scope.status = "...";
        startprogress();
        bluetoothSerial.list(function(d){
            d.forEach(function(element){
                if(element.name == "LAIRD BL600"){
                    bleConnection(element.id);
                }
            });
        }, function(err){
            console.log(err);
            bluetoothSerial.setDiscoveredListener(
                function(d){
                    console.log(d);
                    if(d.name == "LAIRD BL600"){
                        bleConnection(d.id);
                    }
                }
            );
            bluetoothSerial.discoveredUnpaired(
                function(res){
                }, function(err){
                }
            )
        });        
    }```

jordanChalet, 2017-09-18:
Is it possible using this plugin to connect Android phone to Laird BL600 module ?

don, 2017-10-23:
@jordanChalet the BL600 is Bluetooth Low Energy only. This plugin only support Bluetooth Classic for Android.

Try https://github.com/don/cordova-plugin-ble-central which support Bluetooth Low Energy on both iOS and Android

closed 2017-10-23 by don