don/BluetoothSerial #336: SubscribeRawData() stops giving data after some time and after several seconds data appears

Purushothama-VS, 2017-10-11:
Hello , 

I am connecting my IONIC 2 app to CAN2BT  http://www.case-gmbh.de/, and receiving my CAN data . This works fine  for baud rate 500kbit/s but less than that i.e 250kbit/s the subscribedRawData() stops receiving data after some time. After several seconds again the data starts receiving.

this.bluetoothSerial.subscribeRawData().subscribe((data)=> {
console.log(new Uint8Array(data));
});

I also tried read() function but i get empty data everytime

this.bluetoothSerial.read().then((data)=>{
console.log(new Uint8Array(data));
}

I dont know how to use readUntil() in IONIC2, because i don't know how to specify the delimiter for HEX data am sending form CAN. What we receive is Arraybuffer.

markodayan, 2017-10-20:
I am experiencing the same problem as you, except im connecting to a Zephr Heart Rate Monitor. My reads all return empty data, am also doing my project in Ionic. Will advise you if I find out whats wrong, I really hope someone has a solution.

don, 2017-10-23:
@Purushothama-VS you didn't say if you're on iOS (Bluetooth LE) or Android (Bluetooth Classic). In either case subscribing is usually better than read. Check the docs for the CAN2BT. It's possible you need to occasionally send data from the phone to the CAN2BT to keep the connection alive.

@markodayan generally on heart rate monitors you subscribe to receive data when it's available. Also when you get data in a ArrayBuffer it looks empty when you log it. You need to create a new DataView or typed array e.g. Uint8Array.

closed 2018-05-02 by don