don/BluetoothSerial #337: arduino 101

rabee2050, 2017-10-14:
Hi Don,

Many thanks for your cordova plugin, it is perfect and very easy to use.

I was able to build iPhone and android app to control arduino, and here is the link of my app on both store:
iOS- https://itunes.apple.com/us/app/arduino-bluetooth-kit/id1172121838?mt=8
Android- https://play.google.com/store/apps/details?id=com.tatco.bluetooth

My app is working perfectly with: HM-10, Adafruit, Redbear lab and a lot of other modules.

Now i am working to support Arduino 101 in my app, the issue is i was able to send data to Arduino 101 but i tried hard to send data from Arduino 101 to the phone with no luck.

Please do have any example or you know anything to sort this out.

Many thanks
Rabee

rabee2050, 2017-10-22:
Hi,

I have solved the issue, just by send the delimiter '\n' after the data.

For any people who are interested in Arduino 101 board, please check my app on the app store:
https://itunes.apple.com/us/app/arduino-101-kit/id1298510087?mt=8

closed 2017-10-22 by rabee2050

don, 2017-10-23:
@rabee2050 if you want to support Android and iOS for the Arduino 101 check out https://github.com/don/cordova-plugin-ble-central

rabee2050, 2017-10-24:
@don i have used BluetoothSerial and one the predefined characteristic in your library and it worked perfect, anyhow thanks again for your great plugin and for your reply.