don/BluetoothSerial #339: Bluefruit LE Connect - iOS 11.0.3 - Error: Bluetooth is currently powered off

srmo, 2017-10-26:
Hi, I'm crossposting from https://github.com/adafruit/Bluefruit_LE_Connect_v2/issues/29
as your repository popped up in my search for the error mentioned above.
Not sure where I should start. I hope that your repo and the one from adafruit is basically the same.

I just started to receive this error when opening Bluefruit LE Connect.
Bluetooth is currently powered off Bluetooth is enabled on the iPhone and it was working before.
To be honest, I don't know if this only happened since a recent iOS update.

don, 2017-10-26:
This library will connect to Adafruit’s hardware but AFAIK it is not used in Adafruit’s app. 

If you’re using BluetoothSerial in your app and seeing errors please include more info on software versions and how to duplicate.

closed 2017-10-26 by don

srmo, 2017-10-26:
Thanks for the clarification! Seems to have been a ios hickup which was sadly solved by a reboot. Thanks for bothering!