don/BluetoothSerial #341: Can not reconnect after restarting BLE device

tiandongmao, 2017-11-17:
Hello,

I tried to connect my cordova app to a BLE device, and have a reconnection issue. The repro step is as follows:

1. Connect app to BLE device as usual, they can talk to each other successfully.
2. Reset the BLE device (power off and then power on).
3. "ble.disconnect" from the app, succeed.
4. "ble.scan" from the app, succeed,
5. "ble.connect" from the app to the BLE device, succeed.
6.  "ble.startNotification" from the app, got the following exception
          Attempt to invoke virtual method 'java.util.List 
          android.bluetooth.BluetoothGattService.getCharacteristics()' on a null object reference
7. console.log(JSON.stringify(device, null, 2)); print the following info:
    {
      "name": "SWKECG-B0B448DAABE8",
      "id": "B0:B4:48:DA:AB:E8",
      "advertising": {},
      "rssi": -48,
      "services": [],
      "characteristics": []
    }
    Looks like the "characteristics" is empty and hence the exception. 
8. After this exception, I have to reset the bluetooth in my phone to reconnect to the BLE device successfully. 

I also tried to use native Android BLE, and the app can reconnect to the BLE device after around 5 seconds without the need to reset bluetooth. 

Could you please help me figure out how to reconnect without reset Bluetooth? Thanks a lot!

romanlutai, 2017-12-27:
Hi! I believe you have mistaken the repository. 
`ble.` library is [here](https://github.com/don/cordova-plugin-ble-central) but now you are at the BluetoothSerial

don, 2018-05-03:
@tiandongmao, this looks like a https://github.com/don/cordova-plugin-ble-central issue (as @romanlutai mentioned.) If it's still a problem, please create an new issue in that repo.

closed 2018-05-03 by don