don/BluetoothSerial #348: iOS OBD2 (ELM327) Bluetooth support

GPlay97, 2017-12-14:
Hey,

just a little question to ensure that it will work on iOS:
Currently I have an android app using your plugin to communicate over bluetooth with an ELM327 OBD2-Dongle.
It just work's perfectly.
Now I want to do the same with iOS - is this possible? Any limitations? I have read, that iOS doesn't support this kind of communication for obd2?!

Thanks in advance!

romanlutai, 2017-12-27:
[OffTop]
I don't know an answer for your issue, sorry.
But I have issues with scanning for Bluetooth devices. 
I have checked your repository, but haven't seen you using `bluetoothSerial.discoverUnpaired()` in your app.
Have you been using that function? Did it work?

GPlay97, 2018-01-08:
Hello,

sorry for the late response.

No, I'm not using discover unpaired yet.
Currently I'm using list function which lists bonded (paired) devices.

I think to make it work, you need a callback (also error to ensure that nothing is wrong).
You can also set up a discover callback as far as I've understood: https://github.com/don/BluetoothSerial#setdevicediscoveredlistener