don/BluetoothSerial #350: Communications with arduino (HC-06 BT module) partially working

I-Connect, 2017-12-21:
Hi,

I have created my very first app with phonegap using this plugin.

I want to communicate between my S8+  and an arduino mega that has a HC-06 BT module.

I used your chat example as start. I am receiving data on the arduino that is sent from the app. But not receiving data when sent from the arduino to the app. (I tried the same using a different app from the app store; "Bluetooth Terminal" and this works both ways so the issue does not seem to be in the arduino or HC-06 module.

Could you let me know if this should work and if yes a hint what I could be doing wrong?

Thx!!
Jeroen

nebnes, 2018-01-05:
I have the same issue. Have you found a solution ?

I-Connect, 2018-01-06:
HI,

I ended up using this app: https://github.com/1oginov/Cordova-Bluetooth-Terminal as bases.

Regards,
Jeroen

closed 2018-05-02 by don