don/BluetoothSerial #351: Android does not scan for new devices.

romanlutai, 2017-12-26:
Hello,
I am using this combination:
`    bluetoothSerial.setDeviceDiscoveredListener(function (device) {
      console.log(device.id);
      btListHTML.insertAdjacentHTML('beforeend',`<li onclick="bt.pair('${device.name}','${device.id}')">${device.name}</li>`);
    }, function(failure){console.log(failure);} );
    bluetoothSerial.discoverUnpaired( function(devices){
      bluetoothSerial.clearDeviceDiscoveredListener();
    }, function(failure){console.log(failure);} );`
To scan bluetooth devices on Android phone.

Though, seems like either `setDeviceDiscoveredListener()`, or `discoverUnpaired()` are not doing anything. They are not returning failure or information about the device, and they do not see bluetooth devices, even though, the phone itself can see unpaired devices.

Moreover, other functions from the library seems like working fine: before the code written above, the app is succesfully using `bluetoothSerial.isEnabled()`, `bluetoothSerial.enable()`, and `bluetoothSerial.list()`.

What could be the issue?
Br, Roman

andersborgabiro, 2018-02-10:
Do you request BLUETOOTH_ADMIN permission?

don, 2018-05-02:
Closing old issue, assuming permissions fixed the issue.

closed 2018-05-02 by don

Zoidster, 2021-03-09:
I am running into the same problem and adding the BLUETOOTH_ADMIN permission didn't fix it.
Just as OP said, isEnabled and list are working fine but `discoverUnpaired` and `setDeviceDiscoveredListener` don't emit any data.