don/BluetoothSerial #352: Ionic native bluetooth serial disconnect method

ThauanyCorrea, 2018-01-31:
Boa noite, estou utilizando o código a baixo para desconectar o dispositivo:

let alert = this.alertCtrl.create({
title: ‘Disconnect?’,
message: ‘Do you want to Disconnect?’,
buttons: [
{
text: ‘Cancel’,
role: ‘cancel’,
handler: () => {
console.log(‘Cancel clicked’);
}
},
{
text: ‘Disconnect’,
handler: () => {
this.bluetoothSerial.disconnect();
console.log(‘Disconnect clicked’);
}
}
]
});
alert.present();

i able to print “Disconnect clicked” but no response the device still connected. I have to go to android setting screen bluetooth to unpair the devices. “this.bluetoothSerial.disconnect();” this code look like not working. Any others way i can unpair the device? Please advice

don, 2018-05-29:
This plugin's implementation of [disconnect](https://github.com/don/BluetoothSerial#disconnect) works. I'd suggest writing a small non-ionic app to test connect and disconnect with your app. Make sure you process the callbacks. It's possible the [Ionic wrapper for disconnect](https://ionicframework.com/docs/native/bluetooth-serial/) is broken. If that's the case, submit an issue in their repo, or fix it and submit a pull request to their repo.

fihamzuher, 2018-12-05:
Same problem facing this code. Any updates how to solve ?