don/BluetoothSerial #353: Check if connection still valid

fiyi, 2018-02-19:
I am using the bluetooth plugin with ionic 3 and angular and am testing the app on an Android device. 
My app is set up to write to and read from a bluetooth device. Before a write function is called, I use the isConnected() function to check the status of the connection before writing. The problem is this function returns a valid connection even if there is none (I turn off the BT device to check this) The write function when called also writes to the bt device (without a connection) and calls the success handler of the write method.
In the ionic native bluetooth serial plugin it takes about 4-5 seconds for the bluetooth connection status to be updated by the observable from the connect() method. And if a write method is called in this period I get this unreliable behaviour.

Am I missing something else here? How can I make sure that the write function is writing only if there is a valid connection? 

Here is my code - 

    writeData(data) {
	this.bluetoothSerial.isConnected().then(() => {
		this.bluetoothSerial.write(data).then((success) => {
			console.log('written: ' + data + ' success ' + success);
			alert(data + " Written to device!");
		},
		(error) => {
			alert('error writing: ' + error);
		});
	}, (error) => {
		console.log('Operation not completed as device not connected');
		alert('Device is not connected');
	});
    }

don, 2018-05-02:
Don't check the connection state. Once you connect, you'll get a success callback. Later on, if the device disconnects,

LRonHubs, 2018-05-18:
He's saying if you start your app with your device already connected, there's no way to tell whether it's actually connected or simply in the list of previously paired devices.  This is a huge problem for me as well.

Android will auto-pair to a device that's turned on and nearby, if it has previously been paired to the device. The library lacks functionality in recognizing this type of connection and disconnection.

fiyi, 2018-05-19:
@don Thank you, yes thats what I went with - relying on the Observable.

closed 2018-05-29 by don