don/BluetoothSerial #362: Print QRCode

soesantos, 2018-04-27:
Hello don,
pls advice how to print QRCode using this great plugin!!

thanks
GBU

don, 2018-05-02:
@soesantos this plugin give you serial communication. Printing QR code is probably printer specific. I'm guessing you'll need to send some PCL or control codes to get the printer to do what you want. A lot of people ask about this so please update the issue with more details if you find a solution.

closed 2018-05-02 by don

soesantos, 2018-05-02:
hi don,
i try to print QR code generated to ionic card div (red mark) this is Base64 String of Image  ,

i send code "write" to printer 
"
let PrintData = document.getElementById("SKU")
bluetoothSerial.write(PrintData, success, failure);
"
but the output on the printer is print the string of base64 image not print qr code picture
eg. 
[image]
[image]