don/BluetoothSerial #366: How to actually receive data/subscribe/subscribeRawData with the current API?

gabrielhpugliese, 2018-05-25:
The versions I'm using are:
```
   "@ionic-native/bluetooth-serial": "^4.7.0",
    "cordova-plugin-bluetooth-serial": "^0.4.7",
```

But the docs does not seem to be up-to-date. Is it correct?
What I see in the docs as usage for `subscribeRawData` is `bluetoothSerial.subscribeRawData(success, failure);` but when I try to do that, it won't build with error:
```
[16:04:24]  typescript: src/pages/bluetooth-radio/bluetooth-radio.ts, line: 55
            Supplied parameters do not match any signature of call target.

      L54:  console.log('connected to:', address);
      L55:  this.bluetoothSerial.subscribeRawData((data) => {
      L56:    console.log('Data:', data);
```
Because what I see on `index.d.ts` that `subscribeRawData` returns an Observable and receives no params:
```
    /**
     * Subscribe to be notified when data is received
     * @returns {Observable<any>} returns an observable
     */
    subscribeRawData(): Observable<any>;
```

So, based on this information I tried to build this code:
```

  selectDevice(address: any) {
    this.bluetoothSerial.connect(address).subscribe((data) => {
      console.log('connected to:', address, data);
      this.bluetoothSerial.subscribeRawData().map((data) => {
        console.log('Data:', data);
      });
    });
  }

```

Which prints correctly once inside the subscribe but then `subscribeRawData().map` never calls the callback. What I'm doing wrong?

don, 2018-05-25:
This plugin provides a callback based API. Once you have connected, you can pass a callback to [bluetoothSerial.subscribeRawData(callback)](https://github.com/don/BluetoothSerial#subscriberawdata). This callback is called whenever data arrives. For iOS it'll be when there's a notification from the peripheral. For Android it will be when data is written to the connection.

Ionic Native provides a TypeScript wrapper around this plugin. [subscribeRawData()](https://ionicframework.com/docs/native/bluetooth-serial/#subscribeRawData) should work similar to callbacks but using an observable.  Unfortunately subscribe has 2 meaning here. My API uses subscribe to refer to Bluetooth notifications. And you're also subscribing to Observables.

The [next handler](https://angular.io/guide/observables#defining-observers) for subscribeRawData will get called ever time the plugin receives data. This means it can get called 0 times if you never receive data or many times if you're receiving lots of data.

My serial device is sending text so I use TextDecoder to convert the ArrayBuffer to text.

    const decoder = new TextDecoder('utf-8');
    
    this.serial.connect(device.id).subscribe(
      () => {
        console.log('connected to:', device.id);
        this.serial.subscribeRawData().subscribe(
          data => console.log(decoder.decode(data))
        )
      } 
    )

gabrielhpugliese, 2018-05-28:
Ah, got it. Sorry about my noob-ish. Thank you!

closed 2018-05-28 by gabrielhpugliese

qntbkk, 2019-08-29:
hi there, i am getting an error : Can not read property 'subscribe'  of undefined

function ConnectBluetoothDevice(address) {
	// BEGIN USER CODE
	if(address === ""){
		return Promise.reject(false);
	}
	const decoder = new TextDecoder('utf-8');

	this.bluetoothSerial.connect(address).subscribe(() => {
    	console.log('connected to:', address);
    	this.bluetoothSerial.subscribeRawData().subscribe(
      		data => console.log(decoder.decode(data))
    	)
  	} 
)