don/BluetoothSerial #367: Does not handle large payloads

mikehall314, 2018-05-28:
When using this as a print driver, the connection becomes unreliable if a large volume of data is sent (more than a few kilobytes) at one time.  This results in either the remote device locking up, or garbled data being received.

don, 2018-05-29:
I assume you're doing this on Android? That makes sense I mostly use this for serial communication sending small messages. IIRC the internal buffers are a poorly implemented hack. You might be able to increase a buffer size or make some other change to improve this.

Sebdawg, 2018-09-21:
I've been having the same issue with this plugin. I've used the BTprinter plugin in the past and it could handle the write data just fine. Problem is I want the printer to work on my iPad mini and BTprinter only works on Android. So I know my printer is capable of handling the data I sent it. When you say you might be able to increase buffer size are you referring to the plugin or the printer?

0xC0DEBA5E, 2020-03-17:
I know this issue is around for some time, but today a colleague of mine and I had the same issue. 

The first bytes always were ok, but after 10-70 seconds there were always some bytes missing. After some digging through the android portion of the code we found out that the problem could have something to do with the power management of the android devices we were using (we tested with Samsung Tablets with Android 6 and 8). 

Android devices can enter a lower power state in which e.g. sensor data isn't directly delivered to the application and instead held in hardware registers. 
We suspect that the device beeing in this power-saving state (where the CPU is throttled) leads to the loss of small portions of the data. 
It seems that disabling this lower power state by aquiring a wakelock fixed this problem for us.

We will investigate further on this issue and maybe we can provide a fix in form of a PR.

Keeping the device awake by wakelocks is not uncontroversial, because it drains the battery more than allowing the CPU to go asleep. Maybe we could make this optional, e.g. by adding a parameter to the `subscribeRawData` function? Our first approach was to keep the wakelock active only as long as there is an active subscription to the raw data.

0xC0DEBA5E, 2020-03-27:
Our solutiuon with the wakelock worked on the older device with Android 6 but on the newer tablet with Android 8 this didn't work. 

Instead we found out that android loses data while reading from a bluetooth device if no writes happen for a longer time to this device. 

So our solution was to send some dummy data every few seconds to keep the connection alive.