don/BluetoothSerial #368: Correct way to recieve HEX data

DmitriStadnik, 2018-05-30:
I send to my ionic app data that looks like this: 0xAA 0x81 0x5D 0x00 0xE8 0xBD 0x74 0x10 0x00 0x12
What I recieve on the phone looks like this: ��]�k���]
Most of symbols are already replaced with 0xFFFD symbol and I can't find a way to recieve correct data. 

I am using a weird code like this because it was the only way I could actually make my app recieve any data from the hardware:
```
this.bluetoothSerial.subscribeRawData().subscribe(
    (data) => {  
        this.bluetoothSerial.read().then(
          (data) => { 
            console.log('data = ' + data); // here we already have wrong symbols
            this.parseData(data) 
          });
      },
      (error) => {
        alert(error);
      });
```

don, 2018-05-30:
When you subscribeRawData, observable receives the data. You should not call read() inside of subscribeRawData().

Try this

	this.bluetoothSerial.subscribeRawData().subscribe(
	    (data) => {
	  	// data is an ArrayBuffer, convert it to typed array
	    	var bytes = new Uint8Array(data);
		console.log(bytes);
	        this.parseData(bytes);
	    },
	    (error) => {
	    	alert(error);
	    }
	);

DmitriStadnik, 2018-05-30:
Thanks a lot. It worked
One last question. Is documentation up to date?

don, 2018-06-29:
Yes the documentation is mostly up to date.

closed 2018-06-29 by don