don/BluetoothSerial #372: Should I use this plugin?

zabojad, 2018-07-03:
I'm working on a prototype peripheral. The Chinese provider of the prototype changed it recently to use SPP instead of BLE (I was using the cordova ble central plugin successfully until now). The reason for that is that we have a firmware update feature that is too slow with BLE.

I understand that this plugin supports SPP. However when I try to connect to the device with it, I get: `Device connection was lost`.

I've seen  [this ticket](https://github.com/don/BluetoothSerial/issues/152) which show the same symptoms. However, I am not able to know if the prototype is RFCOMM. How can I know it ?

Below are the advertising data I receive with the cordova ble central plugin:
```
{
  "advertising": {},
  "characteristics": [
    {
      "characteristic": "2a00",
      "properties": [
        "Read",
        "Write"
      ],
      "service": "1800"
    },
    {
      "characteristic": "2a01",
      "properties": [
        "Read"
      ],
      "service": "1800"
    },
    {
      "characteristic": "2a04",
      "properties": [
        "Read"
      ],
      "service": "1800"
    },
    {
      "characteristic": "2a02",
      "properties": [
        "Read"
      ],
      "service": "1800"
    },
    {
      "characteristic": "2a03",
      "properties": [
        "Write"
      ],
      "service": "1800"
    },
    {
      "characteristic": "2a05",
      "descriptors": [
        {
          "uuid": "2902"
        }
      ],
      "properties": [
        "Indicate"
      ],
      "service": "1801"
    },
    {
      "characteristic": "2a07",
      "descriptors": [
        {
          "uuid": "2902"
        }
      ],
      "properties": [
        "Read",
        "WriteWithoutResponse",
        "Notify"
      ],
      "service": "1804"
    },
    {
      "characteristic": "2a19",
      "properties": [
        "Read",
        "Notify"
      ],
      "service": "180f"
    },
    {
      "characteristic": "2a23",
      "properties": [
        "Read"
      ],
      "service": "180a"
    },
    {
      "characteristic": "2a29",
      "properties": [
        "Read"
      ],
      "service": "180a"
    },
    {
      "characteristic": "2a24",
      "properties": [
        "Read"
      ],
      "service": "180a"
    },
    {
      "characteristic": "2a25",
      "properties": [
        "Read"
      ],
      "service": "180a"
    },
    {
      "characteristic": "2a26",
      "properties": [
        "Read"
      ],
      "service": "180a"
    },
    {
      "characteristic": "2a28",
      "properties": [
        "Read"
      ],
      "service": "180a"
    },
    {
      "characteristic": "2a2a",
      "properties": [
        "Read"
      ],
      "service": "180a"
    },
    {
      "characteristic": "2a25",
      "properties": [
        "Read"
      ],
      "service": "180a"
    },
    {
      "characteristic": "ff02",
      "descriptors": [
        {
          "uuid": "2902"
        }
      ],
      "properties": [
        "Read",
        "Notify"
      ],
      "service": "ff00"
    },
    {
      "characteristic": "49535343-8841-43f4-a8d4-ecbe34729bb3",
      "properties": [
        "Write"
      ],
      "service": "ff00"
    },
    {
      "characteristic": "ff01",
      "properties": [
        "WriteWithoutResponse"
      ],
      "service": "ff00"
    },
    {
      "characteristic": "49535343-aca3-481c-91ec-d85e28a60318",
      "descriptors": [
        {
          "uuid": "2902"
        }
      ],
      "properties": [
        "Read",
        "Write",
        "Notify"
      ],
      "service": "ff00"
    }
  ],
  "id": "08:EB:ED:00:00:58",
  "name": "cook expert 000088",
  "rssi": -92,
  "services": [
    "1800",
    "1801",
    "1804",
    "180f",
    "180a",
    "ff00"
  ]
}
```

Is it enought to know if it is RFCOMM?

don, 2018-07-03:
This plugin uses SPP/RFCOMM on Android. For iOS it uses BLE and with some services that act similar to SPP. On Android it's looking for the well known RFCOMM 00001101-0000-1000-8000-00805F9B34FB.

I'd see if you can pair the peripheral with a computer and send serial commands to it with something like CoolTerm or (Serial Monitor](https://chrome.google.com/webstore/detail/serial-monitor/ohncdkkhephpakbbecnkclhjkmbjnmlo). There are some Android apps that do RFCOMM too.

If the device is just faking SPP over BLE (look at service ff00 characteristics ff01 and ff02) you are probably better off with [cordova-plugin-ble-central](https://github.com/don/cordova-plugin-ble-central). If it does BluetoothClassic for Android and BLE for iOS you could modify [BLEDefines.h](https://github.com/don/BluetoothSerial/blob/master/src/ios/BLEDefines.h) to support your device.

zabojad, 2018-07-03:
> I'd see if you can pair the peripheral with a computer and send serial commands to it with something like CoolTerm or (Serial Monitor](https://chrome.google.com/webstore/detail/serial-monitor/ohncdkkhephpakbbecnkclhjkmbjnmlo). There are some Android apps that do RFCOMM too.

So, I've tried:
-> installing CoolTerm and pairing my mac to the prototype but it seems that my mac Bluetooth does not see the proto.
-> I've installed the [SPP manager](https://play.google.com/store/apps/details?id=at.rtcmanager&hl=en_US) app on my android. I can see and connect to the proto.  SPP Manager allows to send text to the connected device. I've tried to send the same commands I was using in BLE as an hexa string (for example "A5OF51" to get the status of the proto) but nothing happens after having sent it...

Here are the instructions I received from the chinese provider to adapt my app to the new proto:
> App program need to be modified:
>    A), Android version:
>    To abandon the use of BLE communication, please use SPP mode to communicate.
>    B), iOS version:
>    Need to modify the UUID used for communication: the following iOS's Write uses api:write without response
> [image]

According to those instructions, it seems to be real SPP, no? How could I know it is real SPP? SPP Manager managed to connect to the device successfully...

> If the device is just faking SPP over BLE (look at service ff00 characteristics ff01 and ff02) you are probably better off with cordova-plugin-ble-central.

I've tried to use `cordova-plugin-ble-central`, but it fails when I call `ble.startNotifications()` and `ble.write()` and I get disconnected right away...

zabojad, 2018-07-03:
I've just tried [Bluetooth spp tools pro](https://play.google.com/store/apps/details?id=mobi.dzs.android.BLE_SPP_PRO&hl=fr) which seems better.

I can connect to the peripheral but I'm not receiving anything after having sent a command...

don, 2018-07-03:
This plugin probably does what you need: Use SPP for Android and SPP-like BLE service for iOS. You'll need to modify the plugin to support your UUID 0xFF00, 0xFF01, and 0xFF02 for iOS. For Android it's possible you need to use a different RFCOMM UUID instead of the default one, but it's more likely you need to find the right commands, prefixes, or other magic to make it work. I haven't been using Bluetooth Classic for a while so I can't recall the details.

zabojad, 2018-08-20:
Hi @don !

Thank you for your answers so far.

I've managed to make it work on Android.

On iOS however, the device is not even listed with the `bluetoothSerial.list` command. I'm investigating further but if you have any hint or idea about that problem, please tell me :)...

zabojad, 2018-08-20:
It's OK, it's working, thanks a lot for your work @don !

https://github.com/don/BluetoothSerial/issues/379

closed 2018-08-20 by zabojad