don/BluetoothSerial #375: Disable authority request dialog android or prevent multiple

GPlay97, 2018-07-23:
Hello,

when calling bluetoothSerial.enable(), when the device has disabled bluetooth, a prompt will ask for authority (see similiar issue here: #209).

How can this be disabled within the android code? And another thing. If it will be called multiple times (e.g. within an interval) - a new prompt /  a new window will be created every time. That is annoying. And the user must accept every window, before he can see the app again.
It should wait for the user prompt acceptance or decline, before creating a new window, shouldn't it?

Thanks in advance

don, 2018-07-27:
@GPlay97 it is by design that Android prompts the user to enable Bluetooth. See the warning in [BluetoothAdapter.html#enable()](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter.html#enable()). The plugin uses [ACTION_REQUEST_ENABLE](https://developer.android.com/reference/android/bluetooth/BluetoothAdapter#ACTION_REQUEST_ENABLE) to enable Bluetooth.

GPlay97, 2018-08-01:
Hi, 
I see. 
Thanks. 

But I think the issue with multiple dialogs is still there. 
Your plugin should track, if there is currently a pending request prompt visible. 
Currently, I need to track this manually in my code with a local variable, which is not that cool.

Furthermore I faced the issue, that, when calling bluetoothSerial.enable() it will put the app in the foreground. This is good if it is not enabled (for the prompt) - but if its enable it should not do that. Background: Running a process in background - if called enable, it should only go to foreground, if disabled. Otherwise it should not do that. Currently if enabled, it is also just doing "nothing" and immediately calling the success callback.

don, 2018-08-01:
Android creates the dialog from the Intent. It's up to your app to track if you need to enable Bluetooth or not.

If you're using the BLE part cordova-plugin-ble-central can watch the Bluetooth state with [ble.startStateNotifications(callback)](https://github.com/don/cordova-plugin-ble-central#startstatenotifications).

closed 2018-08-02 by GPlay97