don/BluetoothSerial #379: My peripheral is not listed from iOS

zabojad, 2018-08-20:
I manage to list, connect, write and subscribe successfully to my peripheral from Android.

From iOS however, the `bluetoothSerial.list(success, failure);` call returns `[]`. What could be the reason of that?

Note that I can see the peripheral in the Bluetooth settings of iOS...

zabojad, 2018-08-20:
OK, I think I've found why but I still cannot fix it.

I've added in BLEDefines.h and BLE.m the service and characteristics UUID my peripheral uses.

`BLEDefines.h`
```
#define MYPERIPHERAL_SERVICE_UUID                         "FF00"
#define MYPERIPHERAL_CHAR_TX_UUID                         "FF01"
#define MYPERIPHERAL_CHAR_RX_UUID                         "FF02"
```
`BLE.m`
```
redBearLabsServiceUUID = [CBUUID UUIDWithString:@RBL_SERVICE_UUID];
    adafruitServiceUUID = [CBUUID UUIDWithString:@ADAFRUIT_SERVICE_UUID];
    lairdServiceUUID = [CBUUID UUIDWithString:@LAIRD_SERVICE_UUID];
    blueGigaServiceUUID = [CBUUID UUIDWithString:@BLUEGIGA_SERVICE_UUID];
    myPeripheralServiceUUID = [CBUUID UUIDWithString:@MYPERIPHERAL_SERVICE_UUID];
    hm10ServiceUUID = [CBUUID UUIDWithString:@HM10_SERVICE_UUID];
    hc02ServiceUUID = [CBUUID UUIDWithString:@HC02_SERVICE_UUID];
    hc02AdvUUID = [CBUUID UUIDWithString:@HC02_ADV_UUID];
    NSArray *services = @[redBearLabsServiceUUID, myPeripheralServiceUUID, adafruitServiceUUID, lairdServiceUUID, blueGigaServiceUUID, hm10ServiceUUID, hc02AdvUUID];
    [self.CM scanForPeripheralsWithServices:services options: nil];
```

I am still getting:
```
2018-08-20 14:53:15.091742+0200 TestApp[450:258883] Scanning for BLE Peripherals
2018-08-20 14:53:15.092569+0200 TestApp[450:258883] scanForPeripheralsWithServices
2018-08-20 14:53:18.093219+0200 TestApp[450:258883] Stopped Scanning
2018-08-20 14:53:18.093346+0200 TestApp[450:258883] Known peripherals : 0
2018-08-20 14:53:18.093410+0200 TestApp[450:258883] List of currently known peripherals :
2018-08-20 14:53:18.111510+0200 TestApp[450:258883] []
```

zabojad, 2018-08-20:
If I bypass the scan filter, it lists my peripheral so why isn't it capable of listing it when I give it the right service and characteristics UUIDs?

```
[self.CM scanForPeripheralsWithServices:nil options:nil]; // Start scanning
```

```
2018-08-20 14:59:37.162776+0200 TestApp[454:260499] Scanning for BLE Peripherals
2018-08-20 14:59:37.163337+0200 TestApp[454:260499] scanForPeripheralsWithServices
2018-08-20 14:59:37.308141+0200 TestApp[454:260499] didDiscoverPeripheral
2018-08-20 14:59:37.311110+0200 TestApp[454:260499] New UUID, adding
2018-08-20 14:59:37.311179+0200 TestApp[454:260499] didDiscoverPeripheral
2018-08-20 14:59:40.164327+0200 TestApp[454:260499] Stopped Scanning
2018-08-20 14:59:40.164457+0200 TestApp[454:260499] Known peripherals : 2
2018-08-20 14:59:40.164522+0200 TestApp[454:260499] List of currently known peripherals :
2018-08-20 14:59:40.164591+0200 TestApp[454:260499] 0  |  5BE8F773-2460-FB9E-23C1-A6EF35134458
2018-08-20 14:59:40.164652+0200 TestApp[454:260499] ------------------------------------
2018-08-20 14:59:40.164695+0200 TestApp[454:260499] Peripheral Info :
2018-08-20 14:59:40.164746+0200 TestApp[454:260499] UUID : 5BE8F773-2460-FB9E-23C1-A6EF35134458
2018-08-20 14:59:40.164852+0200 TestApp[454:260499] Name : PERIPHERAL01
2018-08-20 14:59:40.164905+0200 TestApp[454:260499] -------------------------------------
2018-08-20 14:59:40.164963+0200 TestApp[454:260499] 1  |  E7DEE7C7-3E4E-6235-C879-7E1F05D62752
2018-08-20 14:59:40.165016+0200 TestApp[454:260499] ------------------------------------
2018-08-20 14:59:40.165058+0200 TestApp[454:260499] Peripheral Info :
2018-08-20 14:59:40.165108+0200 TestApp[454:260499] UUID : E7DEE7C7-3E4E-6235-C879-7E1F05D62752
2018-08-20 14:59:40.170257+0200 TestApp[454:260499] Name : PERIPHERAL02
2018-08-20 14:59:40.170320+0200 TestApp[454:260499] -------------------------------------
2018-08-20 14:59:40.192578+0200 TestApp[454:260499] [{"id":"5BE8F773-2460-FB9E-23C1-A6EF35134458","rssi":127,"name":"PERIPHERAL01","uuid":"5BE8F773-2460-FB9E-23C1-A6EF35134458"},{"id":"E7DEE7C7-3E4E-6235-C879-7E1F05D62752","name":"PERIPHERAL02","uuid":"E7DEE7C7-3E4E-6235-C879-7E1F05D62752"}]
```

zabojad, 2018-08-20:
[Maybe it's because de FF00 service is not advertised by the peripheral before we connect to it](https://stackoverflow.com/questions/27167297/corebluetooth-scanforperipheralswithservices-not-working-with-service-array)?

zabojad, 2018-08-20:
OK, by passing this scan issue, now if I try to write a command, I'm getting:
```
2018-08-20 16:06:47.120822+0200 TestApp[499:278012] subscribeRaw
2018-08-20 16:06:47.124093+0200 TestApp[499:278012] Error in setting notification state for characteristic with UUID ff01 on service with UUID ff00 on peripheral with UUID E7DEE7C7-3E4E-6235-C879-7E1F05D62752
2018-08-20 16:06:47.124384+0200 TestApp[499:278012] Error code was Error Domain=CBATTErrorDomain Code=6 "The request is not supported." UserInfo={NSLocalizedDescription=The request is not supported.}
```

zabojad, 2018-08-20:
My fault (again :D), I've inverted the TX and RX UUIDs.

It works fine!!! Thank you @don for your awesome plugin :D!!!

The only thing I've noticed is a "Invalid callback id received by sendPluginResult" when I disconnect...
```
2018-08-20 16:12:52.615468+0200 TestApp[503:279675] unsubscribeRaw
2018-08-20 16:12:53.117570+0200 TestApp[503:279675] disconnect
2018-08-20 16:12:53.120508+0200 TestApp[503:279675] bleDidDisconnect
2018-08-20 16:12:53.120593+0200 TestApp[503:279675] Invalid callback id received by sendPluginResult
2018-08-20 16:12:53.152853+0200 TestApp[503:279675] Disconnected from: 08:EB:ED:00:00:58
```

I will try to find out why I'm getting it...

Shashank2406, 2018-08-30:
Hey I am getting the same issue, my device is Zebra CS4070, How do i know which service udid it works on ? Its working fine on android, on ios i get list as [ ] 0

insightzclub, 2019-11-22:
hi where in the BLE.m should I add the following code?
"redBearLabsServiceUUID = [CBUUID UUIDWithString:@RBL_SERVICE_UUID];
    adafruitServiceUUID = [CBUUID UUIDWithString:@ADAFRUIT_SERVICE_UUID];
    lairdServiceUUID = [CBUUID UUIDWithString:@LAIRD_SERVICE_UUID];
    blueGigaServiceUUID = [CBUUID UUIDWithString:@BLUEGIGA_SERVICE_UUID];
    myPeripheralServiceUUID = [CBUUID UUIDWithString:@MYPERIPHERAL_SERVICE_UUID];
    hm10ServiceUUID = [CBUUID UUIDWithString:@HM10_SERVICE_UUID];
    hc02ServiceUUID = [CBUUID UUIDWithString:@HC02_SERVICE_UUID];
    hc02AdvUUID = [CBUUID UUIDWithString:@HC02_ADV_UUID];
    NSArray *services = @[redBearLabsServiceUUID, myPeripheralServiceUUID, adafruitServiceUUID, lairdServiceUUID, blueGigaServiceUUID, hm10ServiceUUID, hc02AdvUUID];
    [self.CM scanForPeripheralsWithServices:services options: nil];"

dionisio28, 2021-05-07:
@zabojad In which part of the code is this excerpt placed?

totaldevservices, 2021-06-04:
Hi @zabojad, can you help us with where exactly you made your changes? 

I followed what you have commented but so far, I'm getting no devices in iOS. Android works perfectly by the way.

zabojad, 2021-06-07:
Hi @totaldevservices, I just replied to you by email...