don/BluetoothSerial #385: the plugin does not work on ios

miguelcs6, 2018-10-17:
In android works correctly, but for a strange reason I can not connect my device in ios, I have already added permissions in info.plis (Privacy - Bluetooth Peripheral Usage Description), I am trying to use bluetoothSerial.list (connectSuccess, connectFailure); and can not find any device, the features of the ipad is as follows:

iPad 5
iOS 12.0.1

Cordova version is 7.1.0
Version of ionic is 4.1.1

Note: in android if it works and it is the same code

miguelcs6, 2018-10-17:
Note: the version plugin cordova-plugin-bluetooth-serial is ^0.4.7

miguelcs6, 2018-10-17:
Note: NodeJS 6.11.2

daniellizik, 2018-11-19:
Same for me on:

- os x 10.13.6
- xcode 10.1
- cordova 8.0.0
- cordova-ios 4.5.5

I got it working with these steps:

- delete the entire `platforms/ios` folder
- rebuild your ios platform with [`cordova build ios --buildFlag='-UseModernBuildSystem=0'`](https://stackoverflow.com/questions/52281848/cordova-build-ios-error-archive-not-found-at-path-path-to-myapp-xcarchive)
- run ios with `cordova run ios --buildFlag='-UseModernBuildSystem=0'`

SandroMoretti, 2019-06-27:
hey guys @patrickpaky  @daniellizik @miguelcs6 
Did you found another alternative?

I'm facing the same issue. You are using another plugin? Thanks.

SandroMoretti, 2019-06-27:
Always returning empty array on ios

DBOST10, 2019-07-14:
I have this issue too. It's always returning an empty array on iOS, on android is working fine.

patrickpaky, 2019-07-15:
> hey guys @patrickpaky @daniellizik @miguelcs6
> Did you found another alternative?
> 
> I'm facing the same issue. You are using another plugin? Thanks.

@SandroMoretti  I just use another plugin for IOS device.
https://github.com/randdusing/cordova-plugin-bluetoothle

hope this helps you.

s00500, 2019-12-30:
Seems like I just ran into this as well...

Maevus, 2020-01-22:
bluetoothSerial.list() also resolves to an empty array for me using "^4.7.0" on iOS. 
Works perfectly on Android.

I have the following log in Xcode using an iPhone 8 (iOS 13.3) with available peripherals and 1 connected bluetooth device.

```
Scanning for BLE Peripherals
scanForPeripheralsWithServices
Stopped Scanning
Known peripherals : 0
List of currently known peripherals :
```

Ionic Info

```
Ionic:

   Ionic CLI          : 5.4.13 (/usr/local/lib/node_modules/ionic)
   Ionic Framework    : ionic-angular 3.9.2
   @ionic/app-scripts : 3.2.4

Cordova:

   Cordova CLI       : 9.0.0 (cordova-lib@9.0.1)
   Cordova Platforms : android 7.1.4, ios 5.1.1
   Cordova Plugins   : cordova-plugin-ionic-keyboard 2.2.0, cordova-plugin-ionic-webview 4.1.3, (and 15 other plugins)

Utility:

   cordova-res : 0.8.1
   native-run  : 0.3.0

System:

   Android SDK Tools : 26.1.1 
   ios-deploy        : 1.10.0
   ios-sim           : 8.0.2
   NodeJS            : v12.14.0 (/usr/local/bin/node)
   npm               : 6.13.4
   OS                : macOS Catalina
   Xcode             : Xcode 11.3.1 Build version 11C504
```

don, 2020-01-23:
iOS and Android work different with this plugin

- Android uses Bluetooth Classic
- iOS uses Bluetooth Low Energy

It's unlikely that the same Bluetooth device will work on both iOS and Android with this plugin. You can't easily support Bluetooth Classic on iOS without joining the MFi program. Let me know if you have suggestions on how to make the documentation clearer.

https://github.com/don/BluetoothSerial#bluetooth-serial-plugin-for-phonegap

closed 2020-01-23 by don

Maevus, 2020-01-24:
Hi Don,

Thanks very much for getting back and for all your hard work on this plugin. Yes on closer inspection of the code and the readme (apologies!!! :) ) I realised that you had flagged that this aimed at specific devices.

Supported Platforms
...
- iOS with RedBearLab BLE hardware, Adafruit Bluefruit LE, Laird BL600, BlueGiga, or HC-02

could be 

Supported Platforms
...
- iOS connecting to BT devices using RedBearLab BLE hardware, Adafruit Bluefruit LE, Laird BL600, BlueGiga, or HC-02 services

But it is actually very clearly flagged in the Misc section. Thanks again for the help and taking the time to clarify.

s00500, 2020-01-24:
Hm, I have specifically tried to match the Adafruit Service and Characteristics ID and could not make it work, it works fine though with other plugins like this 
https://github.com/randdusing/cordova-plugin-bluetoothle

don, 2020-01-24:
@s00500 This is an old plugin, maybe some Adafruit devices use a different service now? For Bluetooth Low Energy on iOS and Android you can use https://github.com/don/cordova-plugin-ble-central or the other Bluetooth plugin.

manish-katti, 2020-04-01:
@don, thanks for your great support.

I wanted to know is  "https://github.com/don/cordova-plugin-ble-central" plugin can be used as Bluetooth serial communication?

ManthanBhatt, 2021-01-29:
@don 
Thank you very much for creating such a great plugin.
I was also facing the same problem where I'm able to connect the Bluetooth device on android seamlessly but on iOS I'm unable to connect the device as the device is not discoverable.

As per the documentation, it is quite clear that the plugin only supports a few Bluetooth hardware.

I was wondering that is there any way we can support classic Bluetooth on iOS. Or another plugin for iOS which allows us to connect with the Bluetooth device and send data for printing?

SimranRanu, 2021-11-01:
@ManthanBhatt do you find any answers to your question?

ManthanBhatt, 2021-11-01:
@SimranRanu For using classing Bluetooth connection type for iOS you can use this plugin https://www.npmjs.com/package/cordova-plugin-bluetoothClassic-serial.

But for this to work you need to confirm the device you want to connect is enrolled in the MFi program of the Apple devices.

Hope this helps.