don/BluetoothSerial #393: how to use write function ,i need use esc/pos print order

baijygo, 2018-12-29:
[image]

baijygo, 2018-12-29:
I want to complete this

uokivan, 2018-12-31:
@baijygo 
can you tel me how to do that, ok?
Thanks a lot & happy new year

baijygo, 2019-01-02:
```
var hello = "01               膳呗外卖";
      hello += " \n ";
      hello += "王老七";
      hello += " \n ";
      hello += "出餐时间: 1月2日 15:28";
      hello += " \n ";
      hello += "-------------------------";
      hello += " \n ";
      hello +=
        "                                                                                                                                                                                 ";
      hello += "锅包肉";
      hello += " \n ";
      hello += "              *1   20.3";
      hello += " \n ";
      hello += "米饭";
      hello += " \n ";
      hello += "              *1   1.5";
      hello += " \n ";
      hello += "-------------------------";
      hello += " \n ";
      hello += "餐盒费               2";
      hello += " \n ";
      hello += "配送费               3";
      hello += " \n ";
      hello += "优惠券              -5";
      hello += " \n ";
      hello += "-------------------------";
      hello += " \n ";
      hello += "              顾客实付21.8元";
      hello += " \n ";
      hello += "15151996663";
      hello += " \n ";
      hello += "顾客";
      hello += " \n ";
      hello += "中国长安网微纳园";
      hello += " \n ";
      var data = this.stringToBytes(hello);
      console.log(data);
      var self = this;
      bluetoothSerial.connect(
        id,
        function() {
          console.log("Ble  connect success");
          bluetoothSerial.write(
            data,
```
[image]
the picture  is the code printe
the printer write  more less than i want to console
how can i config the printer write all i want to print

uokivan, 2019-01-02:
can u share your code?
and I thank the error is:
-----------
  hello +=
        "   
-----------

baijygo, 2019-01-03:
> ## 你能分享你的代码吗？
> 我感谢错误是：
> ## 你好+ = 
> “

function  stringToBytes  use to change utf-8 to gbk  and the `hello +=
"`  is  test code ,
i think that not problem because i can console all  i  want to print 
[image]
[image]
the console is all  but  then i print is the part ,where i can config the printer print more or the data is part i do not know where have problem

closed 2019-01-08 by baijygo

eki01, 2019-07-02:
@baijygo
how to fixed this problem?