don/BluetoothSerial #395: Bulk Print Issue

wovvtech, 2019-02-11:
First of all thanks for this helpfull plugin. Now my issues is i want to print probably 100 lines at a time for my billing app via bluetooth. yes, your plugin work perfect for two and three print with limited lines but after that it prints half receipt or print few lines only.
so i don't know where the original problem is so i think it is best place to find solution.

shql, 2019-03-06:
I think we are at the exact same problem now. A few lines work without a problem but when printing a full receipt it sometimes works with 100%, sometimes just 80% of the lines. @wovvtech Did you find a solution?

wovvtech, 2019-03-20:
not yet

shql, 2019-03-20:
Thanks. We are working around the problem by splitting the data into chunks. 2 Bytes a time works...

shql, 2019-03-25:
Well, it does. But it slows down due to the connection overhead. It's okay for now but we will be back at this point in the future.

Split the raw data:
```
const raw = 'YOUR PRINTER DATA BLABLABLA';
let parts = [];
parts = parts.concat(raw.split(/(.{2})/).filter(O => O));
```

Send the parts to the printer:
```
const printRequests = [];
for (const part of parts) {
  printRequests.push(this.bluetoothSerial.write(part));
}

Promise.all(printRequests).then(res => {
  connection.unsubscribe();
  this.bluetoothSerial.disconnect();
});
```

This works for us slow but stable as long as the print data is correct.  Hope that helps.

uitalloferreira, 2020-11-10:
> Well, it does. But it slows down due to the connection overhead. It's okay for now but we will be back at this point in the future.
> 
> Split the raw data:
> 
> ```
> const raw = 'YOUR PRINTER DATA BLABLABLA';
> let parts = [];
> parts = parts.concat(raw.split(/(.{2})/).filter(O => O));
> ```
> 
> Send the parts to the printer:
> 
> ```
> const printRequests = [];
> for (const part of parts) {
>   printRequests.push(this.bluetoothSerial.write(part));
> }
> 
> Promise.all(printRequests).then(res => {
>   /** Disconnect from printer etc. **/
> });
> ```
> 
> This works for us slow but stable as long as the print data is correct. Hope that helps.

thanks, this implementation provides me with a fix to the same problem.