don/BluetoothSerial #402: Unable to connect to device

andydean565, 2019-07-17:
trying to connect from an Ionic 4 application on my android device (v9) to an 'HC-06' Bluetooth module on an Arduino but it always throws an error when I try to connect.

'Unable to connect to device'

is the 'HC-06' compatible with this Bluetooth plugin? I have been able to connect using an open-source android native app. 

here is my code I'm using to connect :

`this.bluetoothSerial.connect(address).subscribe(success => {
      console.log('connected : ' + success);
    }, error => {
      console.log('error connecting : ' + error);
    });`

kingkarki, 2019-09-19:
I connected Arduino Bluetooth easily. Here is my full code. I hope this will be helpful to you. 
```

import { Component } from '@angular/core';
import { NavController } from 'ionic-angular';
import { BluetoothSerial } from '@ionic-native/bluetooth-serial';
import { Platform } from 'ionic-angular';

@Component({
  selector: 'page-home',
  templateUrl: 'home.html'
})
export class HomePage {
  connected = false;
  devices;
  relay = [
    {
      id: 1,
      type: false,
    },
    {
      id: 2,
      type: false,
    },
    {
      id: 3,
      type: false,
    },
    {
      id: 4,
      type: false,
    },
    {
      id: 5,
      type: false,
    },
    {
      id: 6,
      type: false,
    },
    {
      id: 7,
      type: false,
    },
    {
      id: 8,
      type: false,
    }
    
  ];
  compareWithFn = (o1, o2) => {
    return o1 && o2 ? o1.id === o2.id : o1 === o2;
  };

  constructor(public plt: Platform, public navCtrl: NavController, private bluetoothSerial: BluetoothSerial) {

  }
  ionViewDidLoad() {
    this.plt.ready().then(() => {
      this.bluetoothSerial.isEnabled().then((s)=>{
        console.log('bluetooth is enabled...');
        this.list();
      }, (f)=>{
        console.log('bluetooth is not enabled. i m trying to enable');
        this.bluetoothSerial.enable().then((s)=>{
          console.log('bluetooth enabled via api');
          this.bluetoothSerial.available().then((s)=>{
            console.log('bluetooth is avaliable');
  
           this.list();
  
  
          }, (f)=>{alert(f)});
            console.log('bluetooth is not avaliable');
  
        }, (f)=>{
          console.log('cannot  enable bluetooth via api');
  
        });
  
      });

    });

   

    // this.bluetoothSerial.isConnected().then((s)=>{alert('connected')}, (f)=>{

    // });

  }

  on_off(val){
    
    let data = val.id;

    if(!val.type){
      data = val.id * 10;
     
    }else{
      data = val.id;
    }

 let value = data.toString();
    this.bluetoothSerial.write(value).then((s)=>{alert(s)}, (f)=>{alert(f)});
    

  }

on(value){
	this.bluetoothSerial.write(value).then((s)=>{alert(s)}, (f)=>{alert(f)});

}

is_avaliable(){
	this.bluetoothSerial.available().then((s)=>{alert(s)}, (f)=>{alert(f)});

}
	
list(){
  
  this.bluetoothSerial.list().then((s)=>{
    this.devices = s;
    console.log('bluetooth list avalible');
    console.log(s);

  }, (f)=>{alert(f)});

}

is_connected(){
  this.bluetoothSerial.isConnected().then((s)=>{alert('connected')}, (f)=>{alert(f)});

}

connect_me(address){
  console.log(address);
  this.bluetoothSerial.connect(address).subscribe((a)=>{
    this.connected =true;
    console.log(a);
  },(f)=>{
    this.connected = false;
    console.log(f);
  })
}

}

```

kingkarki, 2019-09-19:
I am controlling 8 channel relay using Arduino with Bluetooth.

don, 2020-01-23:
HC-06 is Bluetooth Classic and should work on Android as long as the serial (RFCOMM) is enabled. It's possible you need to send some initialization commands to the HC-06 to or pair the HC-06 with your phone

closed 2020-01-23 by don