don/BluetoothSerial #404: Add method readRaw

ruipalma, 2019-08-30:
Add method readRaw(), because the method read() returns a string and the javascript charCodeAt() do not return byte values over 127, so its impossible to work with binary data.

jitendracodehacker, 2019-08-30:
Hi you can get it work done with following way 
this.BTSerialIface.subscribeRawData().subscribe( data => {
      //dont use read . it gives you unit8 byte array
          });

ruipalma, 2019-08-31:
Yes, thanks, that's what i'm doing, but it's not the same thing because in an application that you want:
Question -> Answer; Question ->Answer, with timeouts, error control, and flow management the subscribeRawData callback its called out of context and we need to use some workarounds to manage the flow.
I think that if the methods subscribeRawData and subscribeData exists, its coherent that the methods readRaw and read also exists.

don, 2020-01-23:
subscribeRawData is the right way to do this with the plugin. In the past for more specific flow control, I've written new plugins for the specific devices.

closed 2020-01-23 by don