don/BluetoothSerial #407: subscribeRawData stops receiving the data when scan fast.

dhruvwo, 2019-11-26:
I am trying to read data from bluetooth scanner using 'subscribeRawData ' function everything works fine but when i scan fast, 'subscribeRawData' stopes calling.

Bluetooth Serial Version:-
`"@ionic-native/bluetooth-serial": "^5.14.0"`

My code:-
```
 this.bluetoothSerial.connectInsecure('xx:xx:xx:xx').subscribe((success) => {
            this.bluetoothSerial.subscribeRawData().subscribe(async (success) => {
                let readData = decoder.decode(success);
                console.log('readData =>', readData);
            }, async (error) => {
                console.log('subscribeRawData connect', error);
            });
        }, (error) => {
        });
```

dhruvwo, 2019-11-28:
I'm getting the following error:
2019-11-28 15:29:22.020 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.024 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.054 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.073 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.082 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.101 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.110 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.144 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.150 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.220 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.233 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.237 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.244 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.246 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.255 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.324 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.330 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.334 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.341 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.344 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.352 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.357 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.363 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.387 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.397 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.400 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1
2019-11-28 15:29:22.408 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 0
2019-11-28 15:29:22.442 21306-21306/com.reptimeportal.app D/ViewRootImpl@33fe3e4[MainActivity]: ViewPostIme key 1

don, 2020-01-23:
I'm not seeing an error here. I suspect there is an error, and your code isn't handling it or the Observable closes once the error occurs. I suggest you write the simplest app possible using callback to see if you can track down the problem.

closed 2020-01-23 by don