don/BluetoothSerial #413: iOS: Connect to UART/Serial Port BLE

VIMHaos, 2020-02-10:
Hi,
I can't create connection to device and send command to serial port in iOS, in Android it's work fine.
For connect via Android I used: MAC, for iOS: UUID.

But currently problem it's:
"Could not find peripheral CE8D6722-6A98-EC88-C675-E811384B4453., also bluetoothSerial.list() not return list devices to.

How I can connect to UART/Serial Port Emulation over BLE?

Notice: 
Our used device, not visible on the Bluetooth menu in the IOS settings, since IOS does not support Bluetooth 3.0 at all, and in addition, BT4 Low energy devices to don't show up in the list as well, even if the phone is able to process information coming from those devices. 
So, I understude this device requires conection like as using the nRF connect apps?

VIMHaos, 2020-02-14:
Update:
If I try to get list of devices, method `bluetoothSerial.list()` returned always empty array `[]`

So, next I tested this plugin: https://github.com/don/cordova-plugin-ble-central, and I finaly connected to device, but only after get list all devices `this.ble.scan([],30)`.

But I need communication with serial port, may exists some solution?

michelkuehn, 2020-11-30:
FYI, I'm facing the same issue with iOS & BLE. I'm unable to get devices when I'm calling `bluetoothSerial.list()` on iOS with the serial plugin. The same code is working fine on Android.

Did you find any way to solve your issue? Using https://github.com/don/cordova-plugin-ble-central allow me to retrieve a bunch of devices but I'm unable to identify the right one nor connect to it.

kieranbarlow, 2022-08-05:
Same problem, did you solve?