don/BluetoothSerial #422: Android System WebView Problems with version 83.0.4103.96

MaikStar, 2020-06-03:
Google made some important changes in their Bluetooth stack.
Until version 81.0.4044.138 everything works fine and fast.
Serial Port communication now takes up to 15s per request 

Here is my simple code
`
   function connected(data) {
        console.info("connected");
        console.info(data); // sends OK
        bluetoothSerial.subscribe('\n', subscribe, function (error) {
            console.error(error);
        });
    } 

    function subscribe(data) {
        //  Android System WebView 81.0.4044.138 as fast as possible
        //  Android System WebView 83.0.4103.96 up to 15s per request
        console.log(data);
    } 

        bluetoothSerial.isEnabled(
            function () {
                console.log("Bluetooth is enabled");
                bluetoothSerial.connect("00:06:66:75:AF:BA", connected, function (error) {
                    console.error("not connected");
                    console.error(error);
                });
            },
            function () {
                console.log("Bluetooth is *not* enabled");
            }
        );
`
Are there any suggestions and do some of also have this behavior?
I'm using an Android 6.0.1 and Android 8.1 device.

My workaround is to deinstall the Updates for the Android System WebView.
But this resets my cordova app localStorage.

How can we / you fix this ??

don, 2020-07-21:
I don't have a good answer for you. I haven't used this plugin for a long time. If you find a solution, please post here.

MaikStar, 2020-07-26:
At the moment the only solution is to downgrade the version.
But this couldn't be the only solution.
It would be perfect if you or the community may find a better solution.