don/BluetoothSerial #425: I get an issue that says Native module RCTBluetoothSerial tried to override RCTBluetoothSerialModule.

HunterB81, 2020-06-26:
Any idea on how to fix this? This problem stops me from opening the application. 
[image]

HunterB81, 2020-06-26:
[image]
This is a screenshot of my Main Activity java file, as you can see I have no duplicates so I do not understand why I'm getting that error.

don, 2020-07-21:
I'm not sure what that error is from. RCTBluetoothSerial is not part of this plugin.

levinsousa, 2021-12-14:
@HunterB81 Do you fix?

ProgPhil, 2022-04-08:
I have exact same error, plz give solution if you found it!

ProgPhil, 2022-04-08:
It might simply be because of that line : packages.add(new RCTBluetoothSerialPackage());
I think it auto-adds it now, so adding it again, makes it crash