don/BluetoothSerial #426: Unable to connect to device (Android to Bluegiga BLE 112)

kozakhs, 2020-07-20:
Hello,

thank you for the plugin. I am facing issue with connection from Android to Bluegiga Bluetooth Module. If I call "connect" (or connectInsecure, whatever), it always responds with error: Unable to connect to device.

Piece of my code:

bluetoothSerial.connect(device.id, function(connectedDevice) {
        console.log('Connected! '+JSON.stringify(connectedDevice));
     }, function(failure) {
        console.log('Failed connection: ' + failure); // Here it goes ...
});

I have tried to use your another plugin BLE (https://github.com/don/cordova-plugin-ble-central), but that is not what I need. I need to write "raw" bytes directly to the device using bluetoothSerial.write function without GATT stuff (service UUID, characteristic UUID), so thats why I prefered to use this plugin. However, using the ble-central plugin I was able to connect to the device and using BluetoothSerial plugin I am not.

I will appreciate any help or advice.

Thank you very much for your time.

don, 2020-07-21:
This plugin does not support BLE on Android, you need https://github.com/don/cordova-plugin-ble-central. I'm not sure what you mean by "I need to write "raw" bytes directly to the device using bluetoothSerial.write function without GATT stuff". BLE needs a service and a characteristic. You can implement a write function with https://github.com/don/cordova-plugin-ble-central for Android by using the UUIDs for the default BlueGiga serial service https://github.com/don/BluetoothSerial/blob/master/src/ios/BLEDefines.h#L14-L17

Lucas-Wanat, 2022-08-30:
> This plugin does not support BLE on Android, you need https://github.com/don/cordova-plugin-ble-central. I'm not sure what you mean by "I need to write "raw" bytes directly to the device using [...]

Inform me how to solve when the case is not BLE, but classic bluetooth?