don/BluetoothSerial #427: Bluetooth connection in the background

ABRT29, 2020-07-21:
hello, my bluetooth button only has one button. when I click on this button it emits information. This is very simple.

To know if this one emitted information I am currently doing a loop every 10 seconds which tries to connect (use of connect ()) if it succeeds then the button emitted. Unfortunately it does not work in the background, after 10 minutes it stops working ...

How can I improve my verification process?

Thank you

don, 2020-07-21:
On iOS you need to set your application to run in the background. My BLE plugin documents how to do this on iOS https://github.com/don/cordova-plugin-ble-central#background-scanning-and-notifications-on-ios

This should run in the background on Android, but I'm not sure about the details. There might be some permissions required in the manifest especially on newer devices.

ABRT29, 2020-07-22:
Thank you for your prompt response. I'll use your documentation for iOS devices and see how they react. 

For Android devices, I have the following permissions in the manifest :
```
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
```

in addition to your plugin I use the following plugins to test :
_android permission plugin_ for different permissions
_background mode plugin_ to run in the background, useful in our case ?

My looping connection process every 10 seconds on my bluetooth button is in a service (I am using angular) 

```
      this.queueInterval = setInterval(() => {
        const that = this;
        that.connectButton();
      }, 10000);
```

```
    connectButton(){
      this.bluetoothSerial.connect(this.MacAddress[0]).subscribe(
        device => {
          console.log('Connecting success: ' + JSON.stringify(device));
          this.checkConnected(true, this.MacAddress[0]);
          /////// other code ///////
          clearInterval(this.queueInterval);
        },
        error => {
          console.log('Connecting error: ' + JSON.stringify(error));
          if (error.message === 'Connection failed'){
            console.log('Connection failed');
            clearInterval(this.queueInterval);
            this.checkConnected(true, this.MacAddress[0]);
          }
      });
    }
```

```
    checkConnected(connFailed, macAddress){
      console.log('Connection status: ' + connFailed);
      if (connFailed === false){
        this.bluetoothSerial.disconnect(macAddress ).then(
          success => {
            console.log('Disconnect success: ' + JSON.stringify(success));
          },
          err => {
            console.log('Disconnect error: ' + JSON.stringify(err));
        });
      }
      else{
        this.bluetoothSerial.disconnect(macAddress).then(
          ok => {
            console.log('disconnect success: ' + JSON.stringify(ok));
            this.queueInterval = setInterval(() => {
              const that = this;
              that.connectButton();
            }, 10000);
          },
          err => {
            console.log('disconnect error: ' + JSON.stringify(err));
        });
      }
    }
```

did I understand well ?

ABRT29, 2020-07-26:
Any ideas please ?